cyberbotics/webots #2418: Allow ODE parameter adjustments

Simon-Steinmann, 2020-10-29:
It would be great to have more advanced, deep level access to ode parameters such as these:
http://opende.sourceforge.net/docs/group__world.html

Additionally, being able to switch between quickStep and worldStep would be great.

Implementation suggestion:
Add "advanced ODE settings" field to the "worldInfo" node.

bkpcoding, 2021-03-04:
Hey, would like to contribute to this feature, well I do have time now as I am free this month and went through most of the documentation of webots and tried out the tutorials, and even set up my development environment to start contributing to some features or enhancement, will be happy to know that my time in these days will be helpful for the webots community and in turn I will get to learn something new and get to know you all. To give some background I have some good knowledge in C++(even went to the CPP coding style for contributing to webots) and know a bit of control engineering, ROS, Robotic manipulation, and perception but I do lack some software development skills.
     So if you think I can handle this issue please let me know how to start, and if not please do suggest some resources that I can go through and some other features or enhancement that I can contribute to and that will helpful for the community. Thank you for reading my comment though, please do reply if you some ideas, I am pretty sure that I will work to the best of my ability to meet the expectations.

omichel, 2021-03-04:
I am not sure we really need this feature. Do you have a real use-case for that?
If not, I believe you should consider contributing to a different issue.

lukicdarkoo, 2021-03-04:
@bkpcoding, I guess, a good way to start contributing is fixing the bugs:
https://github.com/cyberbotics/webots/issues?q=is%3Aopen+is%3Aissue+label%3Abug+no%3Aassignee

Or, resolving issues with the `good first issue` tag:
https://github.com/cyberbotics/webots/issues?q=is%3Aopen+is%3Aissue+no%3Aassignee+label%3A%22good+first+issue%22

bkpcoding, 2021-03-04:
Yeah sure, I will try to fix some bugs actually was confused about contributing to feature or fixing bugs but chose the former so that I can start from scratch, but I believe that I will be able to contribute to them at some later stage. Thanks though for such a fast reply.

closed 2021-11-15 by BenjaminDeleze