cyberbotics/webots #2426: Performance issue

Acwok, 2020-11-02:
**Describe the Bug**
I notice a drop in performance since commit: https://github.com/cyberbotics/webots/commit/21dcb0dd8545cb7530a7f97933eceefb99f6819d

**Steps to Reproduce**
Here is a world for reproducing the issue: [test_simplified.zip](https://github.com/cyberbotics/webots/files/5476624/test_simplified.zip)
1. open the world with with Webots R2020b
2. start the simulation and check for the speed (11x on my computer)
3. open the same world with Webots build from commit 21dcb0dd8545cb7530a7f97933eceefb99f6819d
4. start the simulation and check for the speed (0.05x on my computer)

**Expected behavior**
The speed should be the same (or even faster with :wink:)

**Screenshots**
Here is a screencast of my simulation using commit 21dcb0dd8545cb7530a7f97933eceefb99f6819d. With R2020b the simulation runs 0.28x even when the arm grab the hexagonal tile. In the video, the speed is twice slower and even more when the arm grab the tile.
https://youtu.be/59HirFdyYnU

**System**
Custom docker based on Ubuntu 18.04 with Webots built from sources

omichel, 2020-11-02:
Just a quick check: did you compile the develop branch with "make release" or "make debug"?

Acwok, 2020-11-02:
Just make

Acwok, 2020-11-02:
Using make release didn't improve the performances

omichel, 2020-11-02:
Yes, make defaults to "release".

PR #2434 (2020-11-05): Fixed performance issue
Fix #2426.
closed 2020-11-05 by omichel