cyberbotics/webots #2428: Running python controller with Anaconda environment

dwicke, 2020-11-03:
Hi,

I'm new to using webots (R2020 b revision 1)but I'm trying to use it with a Mac (10.15.7) with python installed in an Anaconda environment.  I've installed python3.8 as described in the documentation (via Python website).  So I have the Python dynlib in /Library/Frameworks/Python.framework/Versions/3.8 (which seems necessary and isn't included in anaconda).  However, I'm not sure how I can use libraries I've installed in my conda environment to run my controller.  Currently I get an error (see below).  Help would be appreciated.

**Documentation page(s) concerned**
https://cyberbotics.com/doc/guide/using-python?tab-os=macos#macos-installation

**What is the problem with the current documentation?**
Current documentation does not show how to install when using an anaconda version of python.  If Anaconda is not supported then this should also be pointed out.

Currently when trying to use python from within a conda environment I get an error:

```
/Users/dwicke/anaconda3/envs/webots/bin/python /Users/dwicke/Documents/webots/my_first_simulation/controllers/epuck_go_forward/epuck_go_forward.py

Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)
```

**Describe how you would fix it**
Add documentation if I new how to do so.

From my attempts I've been unsuccessful at solving this.  If anyone knows how, I am very interested in learning how.  Let me know if you need additional info.

Thanks,
Drew

omichel, 2020-11-04:
To use Anaconda Python, you need to re-compile the Python wrapper using Anaconda. The procedure is explained here: https://cyberbotics.com/doc/guide/using-python?tab-os=macos#use-an-alternative-python-version. I will try to make this documentation clearer about the fact that Anaconda Python should be setup this way.

closed 2020-11-04 by omichel

chenwthu, 2022-04-01:
I am using Webots R2022a on an Apple M1 MBP, with python3.8 from Anaconda (x86_64 platform). When I recompile the python wrapper via `arch -x86_64 make release`, it seems successful, but segmentation fault occurs when I run `import controller` from an external controller.

chenwthu, 2022-04-02:
> I am using Webots R2022a on an Apple M1 MBP, with python3.8 from Anaconda (x86_64 platform). When I recompile the python wrapper via `arch -x86_64 make release`, it seems successful, but segmentation fault occurs when I run `import controller` from an external controller.

I finally linked Anaconda Python executable file to `/Library/Frameworks/Python.framework/Versions/Current/python` instead of recompiling the wrapper so that Webots could find it. Also, the path should be filled in Webots Preferences.

2black0, 2024-05-25:
> I finally linked Anaconda Python executable file to `/Library/Frameworks/Python.framework/Versions/Current/python` instead of recompiling the wrapper so that Webots could find it. Also, the path [...]

can you share example code to this?

PR #2429 (2020-11-04): Improve Python guide
Fixes #2428.