cyberbotics/webots #2435: From Fedora Webots doesn't boot at all:

brettle, 2020-11-04:
From Fedora Webots doesn't boot at all:
```
[DavidMansolino@localhost ~]$ webots
Info: Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Warning: Could not find the Qt platform plugin "wayland-egl" in ""
Fatal: This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb.

/snap/webots/6/usr/share/webots/webots: line 51:  4091 Aborted                 (core dumped) "$webotsHome/bin/webots-bin" "$@"
```

_Originally posted by @DavidMansolino in https://github.com/cyberbotics/webots/issues/912#issuecomment-535403460_

stefaniapedrazzi, 2020-11-09:
As documented in the User Guide Webots is ensured to run only on Ubuntu:
https://www.cyberbotics.com/doc/guide/system-requirements#system-requirements

But we will evaluate if this issue is easy to fix (by adding the missing plugin).

brettle, 2020-11-11:
This appears to be a bug in the way snaps interactive with the font caches on some systems. The workaround which worked for me is to remove and rebuild my font caches with the following:

```
sudo rm /var/cache/fontconfig/*
rm ~/.cache/fontconfig/*
fc-cache -r
```
The Snapcraft team is [discussing the issue](https://forum.snapcraft.io/t/snapped-app-not-loading-fonts-on-fedora-and-arch/12484/70) and looking into fixes.

omichel, 2020-12-08:
This is indeed documented [here](https://cyberbotics.com/doc/guide/installation-procedure#troubleshooting). I am closing this issue for now as I believe we cannot do much more.

closed 2020-12-08 by omichel