cyberbotics/webots #2441: Display Python API: support Camera.getImage return value

stefaniapedrazzi, 2020-11-09:
Passing the `Camera.getImage` return value to the `Display.imageNew` function is not working on Python and the controller crashes throwing this error:
```
munmap_chunk(): invalid pointer
```

This is probably because `Camera.getImage` returns a string and `Display.imageNew` doesn't support this format.
But Webots should at least throw a meaningful error or accept the output of the `Camera.getImage` function given that this is a common usage example.

Additionally, you should add an example in the Display documentation page showing how to correctly display a Camera output in Python (and other programming languages).

stefaniapedrazzi, 2020-11-09:
I quickly checked the image format returned by the `wb_camera_get_image` function because I couldn't get enough information from the documentation:
* _C/C++ and Python (Camera.getImage)_: **BGRA**
* _Java_: `Camera.getImage` has format **ARGB** but the Java `ImageRef` applies a conversion from big endian to little endian so that the **BGRA** format has to be passed to the `Display.imageNew` function to display it correctly
    https://github.com/cyberbotics/webots/blob/master/resources/languages/java/controller.i#L327-L339
* _Python_ `Camera.getImageArray`: **RGB**
    https://github.com/cyberbotics/webots/blob/master/resources/languages/python/controller.i#L299-L300

I will also check if I can find why different formats have been chosen for Java and Python `Camera.getImageArray`.

Then, I will check that all the information are written in the documentation and add the example about how to show a Camera image in a Display.

omichel, 2020-11-09:
I believe BGRA format was chosen to improve performance, as it used to be faster than RGBA on old graphics hardware (a long time ago). This is not the case any more and we kept this format for backwards compatibility reasons. The RGB and RGBA format are however more standard (when passed to image libraries such as OpenCV). With the introduction of `wren.js`, we may have to drop the BGRA format and use RGBA natively instead as BGRA is not supported by WebGL. So, it could be that we will have to change Webots to rely on RGBA format only so that we could use the very same shaders in Webots and in `webots.js`.

stefaniapedrazzi, 2020-11-09:
Ok. Anyway there are many consistency issues and worst of all the format is not clearly documented.

The difference in Java doesn't seem to rely on performance but only at the little vs. big endianness and the API should be consistent: we cannot output the **ARGB** format and asking the user to input it specifying that is **BGRA**.

Also in Python I don't understand why the two available functions return data using different format.
OpenCV default format is *BGR*, so the Python Camera.getImageArray returning **RGB** is not optimized to be used with OpenCV (that in any case supports any format). It seems more intuitive to return the exact same format as for Camera.getImage, i.e. currently **BGRA**.

omichel, 2020-11-09:
Yes, OpenCV default format is BGR, but Pillow is RGB...

stefaniapedrazzi, 2020-11-09:
By the way I don't want to change anything other than adding meaningful error messages and improving the documentation that is not complete.

I just wanted to mention the inconsistencies in the API.

omichel, 2020-11-09:
OK. Here is an interesting read about it: https://stackoverflow.com/questions/14556545/why-opencv-using-bgr-colour-space-instead-of-rgb

stefaniapedrazzi, 2020-11-18:
I improved the documentation (#2443) and fixed some bugs (#2443 and #2452).

We can later change the internal format to RGBA instead of BGRA and make it consistent between all the functions that return images (currently only the Python `getImageArray` is returning RGB and all the other functions BGRA).

closed 2020-11-18 by stefaniapedrazzi