cyberbotics/webots #2448: Running Webots MacOS may require administrative privs

brettle, 2020-11-11:
**Describe the Bug**
It looks administrative privs are required to run Webots on MacOS
 
**Steps to Reproduce**
1. Download and open the dmg
2. Drag the app to the Applications folder as instructed
3. Open the app
4. When the dialog saying it "can't be opened because the identity of the developer cannot be confirmed." appears, click "Open"
5. A new dialog requiring the admin's name/password appears. It says "You are overriding your security preferences to install an app from an unidentified developer. Enter an administrator's name and password to allow this."

**Expected behavior**
Admin privs not required.

**System**
 - Operating System: macOS Catalina 10.15.7
 - Graphics Card: Intel HD Graphics 4000

omichel, 2020-11-12:
I believe the administrator of the mac should install Webots and run it once, so that it appears in the list of GateKeeper approved apps. After that any user without administrative privilege should be able to run Webots. More information about it is available [here](https://eshop.macsales.com/blog/57866-how-to-work-with-and-around-gatekeeper/).

brettle, 2020-11-12:
Unfortunately, everything at the link you suggested requires admin privs for the initial run. I was just able to workaround that by downloading the dmg from the terminal with:

```
curl -L -O https://github.com/cyberbotics/webots/releases/download/R2020b-rev1/webots-R2020b-rev1.dmg
```

Using the downloaded dmg, I was able to follow the rest of the install instructions and launch webots successfully.

That workaround might be worth adding to the installation instructions.

omichel, 2020-11-13:
I don't understand why downloading the dmg with curl makes a difference security-wise... Do you have any reference on that? But if it actually does, this is a very good solution. So, to be sure to document the right thing, the procedure you followed is:

1. Open the Terminal application.
2. Type `curl -L -O https://github.com/cyberbotics/webots/releases/download/R2020b-rev1/webots-R2020b-rev1.dmg`
3. From the finder, locate the downloaded dmg file, double-click on it.
4. Move the resulting `Webots` folder into the mac `/Applications` folder.
5. Open the `/Application/Webots/` folder from the finder.
6. <kbd>Ctrl</kbd>-click on the Webots icon, select the **Open** menu item and click the **Open** button.

Is that correct?

brettle, 2020-11-13:
The `curl` workaround is mentioned [here](https://disable-gatekeeper.github.io/) among other places. Basically, Gatekeeper relies on the downloading app to tag the download as from the internet. Browsers do that tagging, but `curl` does not. Since the purpose of Gatekeeper is to prevent unsuspecting users from accidentally installing malware from their browser, this is not an unreasonable state of affairs. Only reasonably sophisticated users know how to use the terminal and they certainly aren't going to do so by accident.

Your instructions are correct except that I moved the `Webots` application (it's not a folder) to `~/Applications` instead of `/Applications` because a non-admin user doesn't have write privs to `/Applications`. And I could then just double-click (no Ctrl-click needed) on the Webots icon in the `~/Applications` folder to run it like any other app.

omichel, 2020-11-13:
Thank you for the explanation. It's very helpful. I will improve the documentation accordingly and let you review it before we merge it to the master branch.

omichel, 2020-11-16:
Fixed in #2449.

closed 2020-11-16 by omichel