cyberbotics/webots #2450: Cannot open source file after restarting Visual Studio in Webots R2020b revision 1

dj4park, 2020-11-15:
**Describe the Bug**
Visual Studio results in serveral "C1083	Cannot open source file: 'C:\ProgramFiles\Webots\resources\languages\cpp\TouchSensor.cpp': No such file or directory" errors after restarting VS.

Initiallly, right after creating a new robot controller using the wizrd and opening the VS solution, everything works fine without any errors.  

**Steps to Reproduce**
1. Create a new controller using the "New Robot Controller..." wizard
2. Click on 'C++', then 'Microsoft Visual Studio'
3. Close Visual Studio
4. Open Visual Studio by going to the directorywhere the *.sln was generated
5. Build Solution 

**Expected behavior**
I expect the controller to work just as it did before when I initially created it using the Wizard in Webots. 

**Screenshots**
[image]

**System**
 - Operating System: Windows 10
 - Visual Studio: Microsoft Visual Studio Community 2019, Version 16.6.2
 - Webots: R2020b revision 1

omichel, 2020-11-16:
I just tried to reproduce this problem on my computer, but it works without any problem.
Did you installed Webots in `C:\Program Files\Webots`?
Can you check that the files searched by Visual Studio are indeed there, for example `C:\Program Files\Webots\resources\languages\cpp\TouchSensor.cpp`?

dj4park, 2020-11-16:
It wasn't installed at `C:\Program Files\Webots` by default. 

I have uninstalled Webots and trying to re-install it. By default, it is trying to install on `C:\Users\USER_NAME\AppData\Local\Programs\Webots`. Should this be changed to `C:\Program Files\Webots`?

dj4park, 2020-11-16:
It is working now after having installed in `C:\Program Files\Webots`. Thank you!

closed 2020-11-16 by dj4park

omichel, 2020-11-17:
This is probably due to the fact that you defined the environment variable `WEBOTS_HOME` to `C:\Program Files\Webots` instead of `C:\Users\<YourUsername>\AppData\Local\Programs\Webots`.