cyberbotics/webots #2467: Sound Effects for Web Simulation

yuchen93, 2020-11-18:
**Is your feature request related to a problem? Please describe.**
I am using webots web simulation for remote human-robot-interaction studies and having sound effects will greatly improve the experience.

**Describe the solution you'd like**
A way to play sound in the web simulation would be great (with API callable in controllers).

**Describe alternatives you've considered**
Alternatively a manual entry point to the streaming js module for event-based sound playing would also work.

Any suggestions for alternatives are very much appreciated!

omichel, 2020-11-18:
You could implement this in a robot window: the robot controller would send a special message to the HTML robot window, like "play:my_sound.wav" and the robot window would use the JavaScript API to play the requested sound. See the [related documentation]( https://cyberbotics.com/doc/guide/controller-plugin#robot-window) and [example](https://github.com/cyberbotics/webots/tree/master/projects/robots/mobsya/thymio/plugins/robot_windows/thymio2).

yuchen93, 2020-11-18:
Thanks a lot for the pointers! I will close this issue for now.

closed 2020-11-18 by yuchen93