cyberbotics/webots #2468: webots --sysinfo throws segmentation fault

stefaniapedrazzi, 2020-11-19:
```
stefi@libeccio:~/develop/webots$ ./webots --sysinfo
System: Linux 4.15.0-123-generic x86_64
Processor: Intel(R) Core(TM) i7-8700 CPU @ 3.20GHz
Number of cores: 12
Warning: QOpenGLFunctions created with non-current context
./webots: line 88:  5152 Segmentation fault      (core dumped) "$webots_home/bin/webots-bin" "$@"
```

There is still a problem with the synchronization of the setup of the OpenGL context.
I tested on develop, but not on master.

omichel, 2020-11-19:
I just tested on my Linux VM and it doesn't crash, neither on the master nor on develop branch.

stefaniapedrazzi, 2020-11-19:
I just remembered that I installed some system updates without restarting the machine.
This could likely be the cause of this error.

closed 2020-11-19 by stefaniapedrazzi