cyberbotics/webots #2480: Pedestrian proto difficult to turn into non-static

omichel, 2020-11-30:
To enable non-static behavior with the [Pedestrian proto](https://github.com/cyberbotics/webots/blob/master/projects/humans/pedestrian/protos/Pedestrian.proto), the users have to add a dummy `controllerArg` value, which is odd and difficult to understand. We should implement a simpler way to do this, like for example add a `mass` field that would enable non-static behavior if the mass is greater than 0, or a boolean static field.

Simon-Steinmann, 2020-11-30:
I dont understand the purpose of having a condition on the controller. If people dont want a controller on it, they can just select "none".  But having a mass field or "enable physics" would be good.

ygoumaz, 2022-11-08:
I don't understand this issue. Is it still relevant?