cyberbotics/webots #2483: staticFriction for HingleJoint doesn't work

kidsjtu, 2020-11-30:
**Describe the Bug**
A clear and concise description of what the bug is.

**Steps to Reproduce**
1. When conducting simulation of manipulators, different "staticFriction" values of a HingleJoint lead to the same motor torques.
2. As shown in the screenshot, staticFriction is set to be 0.5, but the max value of the motor torques is smaller than 0.1.

**Expected behavior**
Fix it.

**Screenshots**
[image]

**System**
 - Operating System: Windows 10
 - Graphics Card: NVIDIA GeForce 1060 3 GB

**Additional context**
no

lkx8421, 2023-09-07:
Is fixed? @omichel

omichel, 2023-09-07:
I don't think so. Did you try the latest version of Webots to see if the bug is still there?