cyberbotics/webots #2492: console message without new line is not showed. console message can affect simulation speed

Thomas-Feldmeier, 2020-12-02:
if a robot does only
   printf("hello world")
this is not printed in console (missing new line like \n or std::endl etc.)

if the robot print every timeStep and webots runs as fast as possible (e.g. FastMode), the console print will be the limiting factor for speed. (e.g. with basicTimeStep 16 and 5 robots who do not print: x200 speed. same with print every timeStep: x17 speed)

both issues can be easily handled by the user (adding a newline or printing less often), but they are unexpected and undocumented and may confuse a lot. (why is my debug message not printed? why is the simulation so slow after i only added some debug messages?)

Hint: WbConsole.cpp:
void WbConsole::handleCRAndLF(const QString &msg) {
...
   if (!line.isEmpty()) {
      if (mIsOverwriteEnabled) {
         ...
         textCursor.insertHtml(line);
      } else
        mEditor->appendHtml(line);
      mIsOverwriteEnabled = endsWithCR;

mEditor->appendHtml is a function in QPlainTextEdit in Qt

it is not called for lines with missing \n

in above scenario, 1 robot is as slow as 5 robots if all are printing
so all the prints wait for Qt to handle them (blocking call)

a new thread could speed up the simulation in FastMode
the main thread should never have to wait for this new thread but still synchronization has to be done

maybe like this: (pseudo-code)

int32 counterA, counterB    // 32 bits, so read/write is atomic on an PC or most 32-bit computers

the main thread is increasing counterA only
the new thread is increasing counterB only

both are indices for an cyclic data array with sufficient size

main thread: if(++counterA % SIZE == counterB % SIZE) error SIZE should be higher or the new thread is not working fast enough
  else   data[counterA] = pointer to string to print

new thread: 
  print all data from counterB to counterA (cyclic)
  counterB = counterA (freeing space in data)
  wait for 16 milliseconds (60 fps)

this way the main thread is not slowed at all by synchronization and the new thread still updates the console fast enough

omichel, 2020-12-02:
I am not sure that a new thread could speed-up the console output...
But I would be happy to be wrong on that.
If you find a good solution to one or both of these problems, we will be happy to test and review your contribution in a PR.

Thomas-Feldmeier, 2020-12-02:
the new thread would do the blocking call, so the main thread is not blocked and has not to wait for Qt any more.

a business man tells his secretary to do things for him. so the secretary has to wait on the telephone, but not the business man.

calling appendHtml() does:
   wait until Qt is ready   (the calling process is sleeping, using little/no CPU)
   append

you don't want your main thread to sleep 0.5ms if you are in FastMode

omichel, 2020-12-02:
That sounds good. I am eager to see if that also works that nicely in practice.

Simon-Steinmann, 2020-12-03:
This may or may not be related, but I encountered a line print issue as
 well. I tried to print a repeating error message on the same line in python using
`print("message", end="\r")`

It just did not print anything.

omichel, 2020-12-08:
@Simon-Steinmann: I believe you should use either `\r\n` (Windows standard) or `\n` (Linux and mac OS standard) to mean "new line". Any other thing may have undefined behavior.

closed 2021-11-15 by BenjaminDeleze