cyberbotics/webots #2496: Webots not detecting dedicated GPU ("X.org" as GPU vendor).

CaiusTSM, 2020-12-02:
**Describe the Bug**
Detection of a dedicated GPU seems to have failed on Manjaro Linux. Gives a warning message reporting "X.org" as GPU vendor.

**Steps to Reproduce**
1. Go to Add/Remove software on Manjaro.
2. Search webots.
3. Build the webots 2020b_rev1-1.
4. Run webots.

**Expected behavior**
This warning does not appear and my AMD dedicated GPU is selected.

**System**
Manjaro Linux x86_64
Kernel: 5.8.18-1-MANJARO
CPU: AMD Ryzen Threadripper 1950X (32) @ 3.400GHz
GPU: AMD ATI Radeon RX Vega 56/64

omichel, 2020-12-03:
Here are several questions to help us better understand the problem, please answer all of them:
1. I assume you installed the snap package of Webots, right?
2. Do the Webots simulations run fast, e.g. using GPU acceleration? In fast mode, you should get a value of x100 or more in the speedometer for a simple simulation with GPU acceleration.
3. If not, are you sure your GPU driver is properly installed?
4. What do you get when you type [glxinfo](https://manpages.ubuntu.com/manpages/focal/man1/glxinfo.1.html) in the console?
5. Can you run other GPU accelerated applications, such as [glxgears](https://manpages.ubuntu.com/manpages/focal/en/man1/glxgears.1.html)?
6. If so, are they running fast, e.g. obviously using GPU acceleration?

CaiusTSM, 2020-12-05:
1. It was an AUR package so that may be a problem.
2. I get about x50 on the Boston Dynamics Spot Demo project and x400 for the Lego Mindstorms Rover (in fast mode).
3. NA
4. From running the command I found some interesting bits:
```
server glx vendor string: SGI
server glx version string: 1.4
...
OpenGL vendor string: X.Org
OpenGL renderer string: Radeon RX Vega (VEGA10, DRM 3.39.0, 5.9.11-3-MANJARO, LLVM 11.0.0)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 20.2.3
...
```
It seems like the vendor string printed by the warning was correct but it treats it as an issue, setting all the settings to low.
5. Yes.
6. Yes.

[Full glxinfo.](https://pastebin.com/MTQu0sBm)

omichel, 2020-12-07:
OK, so it seems Webots is actually taking full advantage of your graphics hardware as it runs fast with respect to GPU. The problem is only on the first launch when it considers that your graphics configuration is below the minimum requirements. You can re-enable the 3D features from the Webots preferences and Webots will not warn you again at the next launch.

The code responsible for detecting capable graphics configurations is [here](https://github.com/cyberbotics/webots/blob/eb8e475d4aab23eacb52457dae7d537e1ce4d195/src/webots/gui/WbView3D.cpp#L1204-L1328). I guess it should be fixed so that your case is considered as a high profile configuration and doesn't cause any feature reduction. The culprit is probably [here](https://github.com/cyberbotics/webots/blob/eb8e475d4aab23eacb52457dae7d537e1ce4d195/src/webots/gui/WbView3D.cpp#L1220). Can you test some changes in there to see how to fix the problem? By the way, what is the exact warning message you get? Can you post a screenshot?

ad-daniel, 2021-07-15:
I'm closing this for now, feel free to reopen it if you have additional issues.

closed 2021-07-15 by ad-daniel