cyberbotics/webots #2528: LensFlare shaders errors

stefaniapedrazzi, 2020-12-08:
When opening the `highway.wbt` sample simulation (or any other world with LensFlare) I get these errors:
```
warning: Could not load shared library symbols for ./gd.so.
Do you need "set solib-search-path" or "set sysroot"?
An error occured during shader compilation/linking!
Vertex shader path: /home/stefi/develop/webots/resources/wren/shaders/lens_flare.vert
Fragment shader path: /home/stefi/develop/webots/resources/wren/shaders/lens_flare.frag
webots-bin: PostProcessingEffect.cpp:108: void wren::PostProcessingEffect::Pass::apply(): Assertion `locationTexture >= 0' failed.
```

It works fine on the master branch.

omichel, 2020-12-08:
I am getting a different warning on Windows. In the Webots console I get:
```
ERROR: Shader compilation failed! Filename: lens_flare.vert
0(6) : error C7614: GLSL ES doesn't allow use of reserved word noperspective
```
Which gives me some hints... I will investigate...

stefaniapedrazzi, 2020-12-08:
Ok, thank you. I wanted to test #2281 directly on the develop branch but this is preventing loading the world.

omichel, 2020-12-08:
Simply removing `noperspective ` from `webots/resources/wren/shaders/lens_flare.vert` and `webots/resources/wren/shaders/lens_flare.frag` seems to fix the problem (although the lens flare doesn't seem to work).

omichel, 2020-12-08:
It seems that fixing this problem is not trivial because OpenGL ES doesn't have access to this `noperspective` functionality.
This would require that we pass an additional parameter to the shader and make changes to the shader.
However, the lens flare is anyhow not working properly (see #939).
I would suggest to disable it for now (by simply removing this noperspective keyword) and fix #939 later on.

omichel, 2020-12-09:
The compilation failure was fixed in #2529.
The lens flare effect (which was anyhow [buggy](#939) is now disabled.

PR #2529 (2020-12-08): Fix lens flare shader compilation failure
Fixes #2528.

This is an emergency patch that fixes the compilation error of the shader. However, with this patch, the lens flare effect doesn't show up any more. The lens flare effect is anyhow broken (see #939). Therefore, I would suggest to fix it there as soon as possible.
closed 2020-12-09 by omichel