cyberbotics/webots #2557: Webots sometimes export nan values in the wrl files

lukicdarkoo, 2020-12-11:
**Steps to Reproduce**
1. Add TurtleBot3 Burger to the world
2. Export it as a wrl file
3. Search for nan occurrences in the exported wrl file (you should find around 100)
4. Try to import it back to Webots
5. Error appears `ERROR: '/Users/lukic/Documents/empty.wrl':994:4272: error: Expected floating point value, found 'nan'.`

**Expected behavior**
Should not export nan values.

**System**
 - Operating System: macOS Big Sur

ShuffleWire, 2021-06-12:
I've run to day (on the master branch) :
webots ~/webots/projects/robots/robotis/turtlebot/worlds/turtlebot3_burger.wbt
Export as VRML97
I found no "nan" in the file

stefaniapedrazzi, 2022-06-07:
VRML export is no longer supported and has been removed in R2022b.
It should be checked on macOS if the same error occurs when exporting the scene in HTML/X3D format.

BenjaminDeleze, 2022-06-14:
I tested the HTML/X3D export on macOS and got no Nan

closed 2022-06-14 by BenjaminDeleze