cyberbotics/webots #2563: hokuyo.wbt makes Webots freezing on Ubuntu 18.04 VM

stefaniapedrazzi, 2020-12-14:
In my Ubuntu 18.04 VM, the `hokuyo.wbt` blocks at 32ms and Webots is completely freezed and I have to manually kill it.
I tested with the snap package but I will now check with other packages too.

omichel, 2020-12-14:
I just tested on Windows again and it works fine.

stefaniapedrazzi, 2020-12-14:
It also works fine on my native Ubuntu 18.04.

omichel, 2020-12-14:
OK, so I remember that it is a known bug with some VM which don't support Lidars because of a bug in reading the OpenGL depth buffer. So, this is not a regression and I am afraid we cannot do much to fix it.

lukicdarkoo, 2020-12-14:
Could it be related to?
https://github.com/cyberbotics/webots/pull/2509

omichel, 2020-12-14:
Which VM are you using? VMware or Virtual Box?

stefaniapedrazzi, 2020-12-14:
VirtualBox

stefaniapedrazzi, 2020-12-14:
I just checked the mainstream snap package available from the snap stable repository and it works correctly.
So the issue seems only related to the latest changes introduced in R2021a.
By the way I confirm I have the same problem with the tarball installation too.

omichel, 2020-12-14:
See #32 for the Lidar problem spotted on VMs and its resolution.

stefaniapedrazzi, 2020-12-14:
I will check.
Just note that all the other Lidar simulations works fine.

stefaniapedrazzi, 2020-12-14:
The issue seems

> Could it be related to?
> #2509

Yes, it seems that the problem is that the value is NaN and here we just filter out INF values:
https://github.com/cyberbotics/webots/blob/develop/projects/samples/devices/controllers/hokuyo/hokuyo.c#L69-L72

Just changing the check to ``f != INFINITY && !isnan(f)`` make it work on my system.

PR #2566 (2020-12-14): Fix Webots freeze on NaN value in hokuyo.wbt
Fix #2563: in the VM some lidar values are NaN causing Webots to freeze when handling the `wb_display_draw_line` function.
However, on my native Ubuntu 18.04, it never occurs that lidar values are NaN.
closed 2020-12-14 by stefaniapedrazzi