cyberbotics/webots #2570: Unable to enable sensor in e-puck obstacle avoidance tutorial with Python

gy2256, 2020-12-15:
**Describe the Bug**
The python based controller for obstacle avoidance appears to have a bug when enabling the sensor. 

The tutorial code is the following

```
from controller import Robot, DistanceSensor, Motor

# time in [ms] of a simulation step
TIME_STEP = 64

MAX_SPEED = 6.28

# create the Robot instance.
robot = Robot()

# initialize devices
ps = []
psNames = [
    'ps0', 'ps1', 'ps2', 'ps3',
    'ps4', 'ps5', 'ps6', 'ps7'
]

for i in range(8):
    ps.append(robot.getDevice(psNames[i]))
    ps[i].enable(TIME_STEP)

leftMotor = robot.getDevice('left wheel motor')
rightMotor = robot.getDevice('right wheel motor')
leftMotor.setPosition(float('inf'))
rightMotor.setPosition(float('inf'))
leftMotor.setVelocity(0.0)
rightMotor.setVelocity(0.0)

# feedback loop: step simulation until receiving an exit event
while robot.step(TIME_STEP) != -1:
    # read sensors outputs
    psValues = []
    for i in range(8):
        psValues.append(ps[i].getValue())

    # detect obstacles
    right_obstacle = psValues[0] > 80.0 or psValues[1] > 80.0 or psValues[2] > 80.0
    left_obstacle = psValues[5] > 80.0 or psValues[6] > 80.0 or psValues[7] > 80.0

    # initialize motor speeds at 50% of MAX_SPEED.
    leftSpeed  = 0.5 * MAX_SPEED
    rightSpeed = 0.5 * MAX_SPEED
    # modify speeds according to obstacles
    if left_obstacle:
        # turn right
        leftSpeed  = 0.5 * MAX_SPEED
        rightSpeed = -0.5 * MAX_SPEED
    elif right_obstacle:
        # turn left
        leftSpeed  = -0.5 * MAX_SPEED
        rightSpeed = 0.5 * MAX_SPEED
    # write actuators inputs
    leftMotor.setVelocity(leftSpeed)
    rightMotor.setVelocity(rightSpeed)

```

I got the error message 
```
Traceback (most recent call last):
  File "e-puck_go_forward.py", line 20, in <module>
    ps[i].enable(TIME_STEP)
AttributeError: 'NoneType' object has no attribute 'enable'
WARNING: 'e-puck_go_forward' controller exited with status: 1.
```

**System**
 - Operating System: Windows 10
 - Python 3.7

lukicdarkoo, 2020-12-15:
I have just tested and it works fine for me. Can you confirm that you use the e-puck robot? If the robot doesn't have the `ps[x]` distance sensor then `robot.getDevice` returns `None` and you get the same error.

gy2256, 2020-12-15:
> I have just tested and it works fine for me. Can you confirm that you use the e-puck robot? If the robot doesn't have the `ps[x]` distance sensor then `robot.getDevice` returns `None` and you get the same error.

Thanks for the reply. I am using the e-puck robot. Here is the output of "show robot window":
[image]

For debugging purposes, I have added the following line in the code
```
psNames = [
    'ps0', 'ps1', 'ps2', 'ps3',
    'ps4', 'ps5', 'ps6', 'ps7'
]

device = robot.getDevice(psNames[0])
print("Device name is : ",psNames[0])
print(device)
```

The console outputs 
```
('Device name is : ', 'ps0')
None
```

It looks like the device is empty in my case. By the way, I am on Webots R2020b.

gy2256, 2020-12-16:
The problem is fixed after I update to the latest Webots R2021a. The python controller works flawlessly now.

closed 2020-12-16 by gy2256

JamalShams, 2021-01-05:
I am facing the same error although i have R2021. What is happening with it? @gy2256 can you tell me how this is solved?

JamalShams, 2021-01-06:
> I have just tested and it works fine for me. Can you confirm that you use the e-puck robot? If the robot doesn't have the `ps[x]` distance sensor then `robot.getDevice` returns `None` and you get the same error.

is `gyro` sensor available in dgi drone? because i am having the `enable` issue although i have R2021a

lukicdarkoo, 2021-01-06:
@JamalShams The DJI drone has `Gyro` sensor. How do you enable it?

JamalShams, 2021-01-07:
Yes the DJI has `Gyro` sensor. i imported it like `from controller import Robot, Gyro`. i used `gyro = robot.getDevice('Gyro')` but in the [documentation](https://cyberbotics.com/doc/reference/gyro?version=R2021a&tab-language=python) says that i have to enable the sensor just after getting it, but when i use the above mentioned command to getDevice; it gives none. i.e. gyro = None. 
Now my problem is how to get that Device (Gyro) so that i can enable it and get its value for my use?

lukicdarkoo, 2021-01-07:
It works fine for me:
```Python
from controller import Robot

robot = Robot()
timestep = int(robot.getBasicTimeStep())
gyro = robot.getDevice('gyro')
gyro.enable(timestep)

while robot.step(timestep) != -1:
    print(gyro.getValues())
```

JamalShams, 2021-01-07:
Thank you . Problem Solved. Thanks alot. i have been working on it for 5 days. thanks again.