cyberbotics/webots #2588: incorrect detection of GPU memory on iMac 2020

Bradford-Miller, 2020-12-19:
**Describe the Bug**
On starting WeBots 2020a for the first time, it generates a warning that my system is below minimum requirements having less than 2Gb of memory. However, "About this Mac" reports an AMD Radeon Pro 5500 XT 8 GB card. (iMac 20,1)

**Steps to Reproduce**
See above

**Expected behavior**
No error and a dawning realization that this is a sufficient graphics card to run Webots uninhibited.

**Screenshots**

**System**
 - Operating System: MacOS Catalina
 - Graphics Card: AMD Radeon Pro 5500 XT 8 GB

**Additional context**
Add any other context about the problem here.

Bradford-Miller, 2020-12-19:
Sorry, that should have read Webots 2021a.

omichel, 2021-01-04:
Could you please post a screenshot of the window you get when going to the "Help" menu, "OpenGL information..."?

omichel, 2021-01-04:
The detection of available GPU RAM doesn't work well in case of AMD cards: it retrieves the currently available GPU RAM, e.g., the memory not currently in use, and not the total GPU RAM. Therefore, if some GPU RAM is already used by another process, Webots may believe that your amount of GPU RAM is way smaller than it actually is. 
It is not possible (or very complicated) to retrieve the total GPU RAM information on macOS in case of AMD GPU.

Given that hardware requirements below the Webots specification correspond now to pretty old machines, I would suggest that we drop these problematic tests in Webots, as they cause more trouble to people with recent hardware than they help people with outdated hardware.

closed 2021-01-05 by omichel

the-raspberry-pi-guy, 2021-01-31:
Just wanted to contribute that this is also a present and annoying issue when using Webots on my MacBook Pro 16 2019. I have a very high spec configuration, but presumably I am suffering from the same GPU AMD issues that you are discussing @omichel :

[image]

I am trying to run on a 4K screen, when I snap the window to only half of the size of the display though I start to get some better performance (ie: a real time factor of 1 and not very much lag)

omichel, 2021-02-01:
This should not affect the performance. 4K resolutions yield anyhow to a lower performance whatever your graphics card and operating system. But it also highly depend on the simulation. Which world file are you running to get such a bad performance? Is that a sample world included with Webots?

PR #2623 (2021-01-05): Fix GPU tests
Fixes #2588.

In addition to fixing #2588, this PR also removes some annoying warnings with recent Intel or AMD graphics cards which support OpenGL very well and should not cause such warnings.

Our [telemetry](https://cyberbotics.com/telemetry) reports the following GPU usage: [gpu.txt](https://github.com/cyberbotics/webots/files/5769060/gpu.txt)