cyberbotics/webots #2594: ICONS don't fit on the 4K screen

Cpy-bit, 2020-12-22:
ICONS don't fit on the 4K screen. They looks very small, and there is no options to resize it.

omichel, 2020-12-22:
Can you please give us more information about your operating system and post a screenshot showing the problem?

Cpy-bit, 2020-12-27:
My operating system : WIN 10
My Screen resolution: 3840*2160
[image]

lukicdarkoo, 2021-01-05:
I can reproduce this on my laptop. It is not related to the high resolution, but to the display scaling (go to `Display > Scale and layout` and set to any number greater than 100).

omichel, 2021-01-06:
Another solution is to define the following environment variable in your system: `QT_ENABLE_HIGHDPI_SCALING=1`

PR #2631 (2021-01-06): Fixed high DPI support on Windows and Linux
Fixes #2594.
closed 2021-01-06 by omichel