cyberbotics/webots #2605: DistanceSensor returns max value when facing certain TexturedBox faces

WillB97, 2020-12-23:
**Describe the Bug**
When using the sonar DistanceSensor with an environment constructed from TexturedBox elements certain areas appear to return the maximum value regardless of the distance from the wall. These areas appear to correspond to one of the triangles on each face.

**Steps to Reproduce**
1. Create a TexturedBox
2. Place a sonar DistanceSensor perpendicular to the created box
3. Move the DistanceSensor parallel to the box along its whole length

**Expected behavior**
The DistanceSensor should return a consistant value corresponding to the distance from the box.

**Screenshots**
[image]

**System**
 - Operating System: Windows 10 (also replicated on Manjaro Linux)
 - Graphics Card: NVIDIA GeForce GTX 1070 8 GB

omichel, 2020-12-23:
Could you upload a super simple simulation showing this wrong behavior?
We need a world file with a super simple robot and a sonar and a very simple robot controller preferably in python that displays the wrong sensor readings.
This would help us tracking down the problem.

WillB97, 2020-12-23:
Here's the stripped down version of our scenario. [TexturedBox-demo.zip](https://github.com/cyberbotics/webots/files/5735896/TexturedBox-demo.zip)

closed 2020-12-30 by ad-daniel