cyberbotics/webots #2608: Using Matlab in Webots

mohakhalili, 2020-12-27:
hi guys

I have a problem with using Matlab in Webots.
When I try to open Webots sample projects for this purpose like nao_matlab or e_puck, it gives the following error. 

[image]

This is while I added MATLAB to  PATH environment variable and also looked at Webots' User guide (Matlab section) and did other things like installing MATLAB MinGW-w64 C/C++ Compiler, but I still get this error.

could you please help me to solve this problem ?

lianne, 2021-09-28:
Hi guys, I still have such a problem on my Mac OS 10.14.6 (Mojave) with Matlab R2019a. I tested both Webots R2021b and R2021b Nigthly Build Rev 1 (version 27 Sep 2021). Matlab starts without problems from the terminal with the command Matlab -nosplash -nodesktop (I have a symbolic link in usr/locan/bin).
Thanks.
 Luigi

claretb, 2022-02-03:
> Hi guys, I still have such a problem on my Mac OS 10.14.6 (Mojave) with Matlab R2019a. I tested both Webots R2021b and R2021b Nigthly Build Rev 1 (version 27 Sep 2021). Matlab starts without [...]

Hi, could you solve the problem? I still have the same issue.

ygoumaz, 2022-02-09:
I can't reproduce this issue on MacOS BigSur 11.6.1 with Matlab r2021b and Webots r2022a. Directly after installation of Matlab and Webots, I get the same error messages as @mohakhalili originally. But after following the Webots user guide on how to add a symlink in "/usr/bin", Webots is capable to launch Matlab and the controllers of the sample projects.

Have you followed the procedure to add the Matlab symlink? What is the version of Matlab and Webots you are using @claretb?

claretb, 2022-02-09:
@ygoumaz My Mac version is Monterey 12.2 and I use Matlab r2021b with Webots r2022a. But I have to say that my Mac has M1 processor. I have followed the symlink guide and added in "usr/local/bin". After you said, I also tried to add symlink in "usr/bin" but it seems Mac does not allow that.

ygoumaz, 2022-02-09:
@claretb @lianne Do you get the same error message as @mohakhalili? Or is it different?

lianne, 2022-02-09:
[image]
here it is @ygoumaz

ygoumaz, 2022-02-10:
I think I found the source of the problem. Apparently the path is not the same when an application is launched from the Finder or from a terminal. On my side it was working because I was launching Webots from a terminal. I assume you were launching the Desktop app from the Finder?

If so, here is a way to add your terminal PATH (which includes /usr/local/bin with the Matlab symlink) to the path of the Finder.

1. Open a terminal and run the following command: 
```
sudo launchctl config user path $PATH
```
2. Reboot your system.
3. (_This step may be optional, but sometimes it was mandatory to solve the problem_) Open a terminal and run the following command:
```
sudo killall Dock
```
4. Now you should be able to open Webots from the Finder and/or the terminal and Matlab should be opened correctly.

If this solves the problem for you two, I'll add this method to the user guide.

omichel, 2022-02-11:
We could maybe implement a new Webots preferences containing the full path to the MATLAB command. The default value for this path could be computed based on the rule explained [here](https://ch.mathworks.com/matlabcentral/answers/175704-where-is-matlab-installed-on-my-machine), privileging the most recent version of MATLAB found on the system. If none is found, it should be left empty and recomputed at each launch of Webots. That would probably save a lot of troubles to users.

lianne, 2022-02-11:
> Yes, it works for me without the optional step. Thank you so much! @ygoumaz

me too, many thanks!!