cyberbotics/webots #2612: Wrong OpenGL version detected

Serafadam, 2020-12-30:
**Describe the Bug**
Hi, I've been trying to run a snap version of Webots on a fresh installation Linux with latest AMD drivers installed. Sadly the simulator won't start, or after a fresh snap installation will start with the welcoming screen, and after that it will quit with the message:

```
webots  

libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: radeonsi
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: radeonsi
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
FATAL: Webots requires OpenGL 3.3 while only OpenGL 1.4 can be initialized.
Please check your GPU abilities and install the latest graphics drivers.
Please do also check that your graphics hardware meets the requirements specified in the User Guide.
```

glxinfo tells me that I currently have OpenGL 4.6 installed:

```
glxinfo | grep OpenGL

OpenGL vendor string: Advanced Micro Devices, Inc.
OpenGL renderer string: AMD Radeon (TM) R9 390 Series
OpenGL core profile version string: 4.6.14756 Core Profile Context 20.45
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.6.14756 Compatibility Profile Context 20.45
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL extensions:
OpenGL ES profile version string: 4.6.14756 Compatibility Profile Context 20.45
OpenGL ES profile shading language version string: 4.60
OpenGL ES profile extensions:
```

Without the drivers, Webots launch properly, but have graphic options limited.

**Steps to Reproduce**
1. Install pro drivers as in https://amdgpu-install.readthedocs.io/en/latest/install-installing.html#installing-the-pro-variant
2. glxinfo
3. Install Webots via Snap
4. Run Webots in terminal
5. See error

**Expected behavior**
Webots working properly

**System**
 - Operating System: Kubuntu 20.04
 - Graphics Card: AMD Radeon R9 390 8GB

stefaniapedrazzi, 2021-03-01:
Do you get the same issue if you install Webots using the tarball package instead of the snap package?

Note that if you run `glxinfo` from a terminal then the environment could be different than the one on the snap container.

stefaniapedrazzi, 2021-03-03:
Could be related to #2496.

Serafadam, 2021-06-22:
Hi, sorry for the long delay,
I tested deb, as well as a tarball package.
The deb shows me the same errors as in snap, but the tarball works perfectly with all features enabled. Now I wonder what is the difference between them

Serafadam, 2021-06-22:
Ok, it seems I found the issue, when running `amdgpu-pro` install some drivers are not installed, when using regular `amdgpu-install` apt works, tarball works, snap does not work.

closed 2021-07-13 by Serafadam