cyberbotics/webots #2619: wb_camera_recognition_disable_segmentation can not find in generic.dll

minji-o-j, 2021-01-04:
**Describe the Bug**
When I Open the project (Didn't run it, just Open), the error occured.
wb_camera_recognition_disable_segmentation can not find in generic.dll

It was okay when I used the version 2020...

**Screenshots**
[image]

**System**
Webbots 2021a

**Additional context**
```
ERROR: 'C:/Users/308_2/minji/aisoccer-3d/protos/soccerfield/SoccerField5v5.proto': error: Wrong order of fields in the instance of PROTO SoccerField: USE nodes might refer to the wrong DEF nodes.
WARNING: WorldInfo > DEF wall-ball ContactProperties: Invalid 'softCFM' changed to 0.001. The value should be positive.
INFO: supervisor: Starting controller: C:\Users\308_2\minji\aisoccer-3d\controllers\supervisor\supervisor.exe
INFO: soccer_robot_v2_GK: Starting controller: python.exe -u soccer_robot_v2_GK.py 0
INFO: soccer_robot_v2: Starting controller: python.exe -u soccer_robot_v2.py 0
INFO: soccer_robot_v2: Starting controller: python.exe -u soccer_robot_v2.py 0
INFO: soccer_robot_v2: Starting controller: python.exe -u soccer_robot_v2.py 0
INFO: soccer_robot_v2: Starting controller: python.exe -u soccer_robot_v2.py 0
INFO: soccer_robot_v2_GK: Starting controller: python.exe -u soccer_robot_v2_GK.py 0
INFO: soccer_robot_v2: Starting controller: python.exe -u soccer_robot_v2.py 0
INFO: soccer_robot_v2: Starting controller: python.exe -u soccer_robot_v2.py 0
INFO: soccer_robot_v2: Starting controller: python.exe -u soccer_robot_v2.py 0
INFO: soccer_robot_v2: Starting controller: python.exe -u soccer_robot_v2.py 0
Error: ������ ���ν����� ã�� �� �����ϴ�.
 (dynamic library)
Error: failed to load C:/Program Files/Webots/resources/projects/plugins/robot_windows/generic/generic.dll library
abort: api/robot.c:347: assertion failed: tag < robot.n_device
abort: api/robot.c:347: assertion failed: tag < robot.n_device
WARNING: 'supervisor' controller exited with status: 1.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
DEPRECATION: Robot.getMotor is deprecated, please use Robot.getDevice instead.
```

stefaniapedrazzi, 2021-01-04:
Did you try to recompile your controller program within Webots R2021a?

minji-o-j, 2021-01-04:
@stefaniapedrazzi 
No I just opened the .wbt file that works well in R2020 version, I just reinstall another version.

stefaniapedrazzi, 2021-01-04:
You should always recompile your controllers with the newer version for major release changes (like for example from R2020b to R2021a) because the backward compatibility with old controller programs compiled with older versions of Webots is not guaranteed.

Please check if this solves your issue or close this report.

closed 2021-01-05 by minji-o-j