cyberbotics/webots #2633: Streaming simulation over ssl

yuchen93, 2021-01-06:
**Describe the Bug**
I am trying to stream a simulation with ssl connection but got an websocket 1006 error.

**Steps to Reproduce**
I started webots simulation on the server with the 'ssl' option in the 'streaming':
`xvfb-run webots --batch --mode=realtime --stdout --stderr --minimize --stream="port=1235;monitorActivity;mode=x3d;disableTextStreams;ssl" ./worlds/<world_file>`

I then used the streaming viewer.html file but changed ws to wss:
`view = new webots.View(document.getElementById("playerDiv"), 0);`
`view.open("wss://"+ip+":1235")`

**Expected behavior**
I expect the simulation streaming to work the same as with ws.

**Screenshots**
[image]

**System**
 - Operating System: Linux Ubuntu 18.04, AWS server

omichel, 2021-01-07:
You also need to add the file `privkey.pem` in the same folder as `fullchain.pem`.

_Note_: to run Webots server over SSL, it is recommend (and easier) to use port rewrite instead of SSL configuration of Webots. This way, you can run Webots server in non-SSL mode and have your Apache server redirect SSL traffic to it, so that from a client perspective, everything works with SSL. This is explained [here](https://cyberbotics.com/doc/guide/web-simulation#port-rewrite).

yuchen93, 2021-01-11:
Thanks! My problem has been resolved.

closed 2021-01-12 by omichel