cyberbotics/webots #2641: The external controller fails if started before Webots in Windows

lukicdarkoo, 2021-01-13:
**Describe the Bug**
If you haven't started Webots after Windows turned on (probably the `\\.\pipe` folder is not created) the external controller fails.

**Steps to Reproduce**
1. Start Windows
2. 
  ```log
  C:\Users\lukic>set WEBOTS_HOME=C:\Users\lukic\AppData\Local\Programs\Webots
  C:\Users\lukic>set PATH=%WEBOTS_HOME%\lib\controller;%WEBOTS_HOME%\msys64\mingw64\bin;%WEBOTS_HOME%\msys64\mingw32\bin;%PATH%
  C:\Users\lukic>set PYTHONPATH=%WEBOTS_HOME%/lib/controller/python38
  C:\Users\lukic>python -c "from controller import Robot; r = Robot()"
  Cannot open pipe file: \\.\pipe\webots_6068_1192306500
  Cannot connect to Webots on pipe: \\.\pipe\webots_6068_1192306500
  ```

**Expected behavior**
The external controller should wait for Webots.

**System**
 - Operating System: Windows 10

omichel, 2021-03-01:
It seems due to the fact that you have an old tmp folder that contains an outdated file (probably due to a previous crash of Webots):
Deleting `C:\Users\<username>\AppData\Temp\webots-6068\WEBOTS_SERVER` should fix the problem.
I am wondering whether the controller should delete this folder in case the pipe file doesn't exists to avoid the crash of the controller.

lukicdarkoo, 2021-03-01:
> I am wondering whether the controller should delete this folder in case the pipe file doesn't exists to avoid the crash of the controller.

Delete or ignore the folder, both sound like good solutions

omichel, 2021-03-01:
I will try to ignore it and display a warning about it to instruct the user to clean it up.

PR #2800 (2021-03-08): Fixed extern controller startup
Fixes #2641.
closed 2021-03-08 by omichel