cyberbotics/webots #2644: Wrong size of robot component device panel

stefaniapedrazzi, 2021-01-13:
In the documentation, the width of the device panel of the robot compontent is too small and the joint limits values are not completely visible.
An horizontal scrollbar is added but this looks bad.
Instead the width should be big enough to display everything and no scrollbar should be visible.

Example:
https://www.cyberbotics.com/doc/guide/bioloid?version=released

[image]

bkpcoding, 2021-02-25:
Is this issue or closed? I am a first-time contributor trying to contribute to webots, if it is open can you help me to proceed or some other issue that I can work on.

omichel, 2021-02-25:
Welcome @bkpcoding!
As you can see, the issue is still open. The code that generates this component is located [here](https://github.com/cyberbotics/webots/blob/master/docs/js/viewer.js#L916).
Otherwise, feel free to pick any of these [beginner issues](https://github.com/cyberbotics/webots/issues?q=is%3Aopen+is%3Aissue+label%3A%22good+first+issue%22).
Finally, this [idea list](https://github.com/cyberbotics/webots/wiki/Google-Summer-of-Code-2021) may also inspire you.

closed 2021-06-03 by BenjaminDeleze