cyberbotics/webots #2650: Low timestep to fix wheel sinking slows down the simulation

anastasiapan, 2021-01-13:
**Describe the Bug**
I am trying to run a simulation with a robot that has a payload about 45kg. First I noticed that wheels are sinking and tried to fix like that:  [https://cyberbotics.com/doc/guide/modeling#my-robot-sinks-into-the-ground-how-to-remove-this-effect](url) as suggested at: [https://github.com/cyberbotics/webots/issues/2645](url). When the robot does not move the wheels do not sink for an extremely low basic TimeStep (0.1). With a time step this low simulation does not run smoothly it looks like it starts very slowly and then the robot is about to fly.

**Steps to Reproduce**
1. Open a simple world and add a robot model  but not above 100 kg (~45-50kg)
2. Change basic TimeStep to 0.1 in the WorldInfo

**Expected behavior**
The robot to drive smoothly without the wheels sinking in the floor

**Screenshots**
Not possible

**System**
 - Operating System: [Windows 10]
 - Graphics Card: [Nvidia Geforce RTX 2060]

**Additional context**
Add any other context about the problem here.

Simon-Steinmann, 2021-01-15:
Can you post your world file? There are many things that can affect that. CFM / SoftCFM is one of the major variables impacting this. Also boundingObject geometry and relative masses. With SoftCFM you have to make sure it is applied to the correct materials.

anastasiapan, 2021-01-17:
> Can you post your world file? There are many things that can affect that. CFM / SoftCFM is one of the major variables impacting this. Also boundingObject geometry and relative masses. With SoftCFM you have to make sure it is applied to the correct materials.

My world file is the same as in my previous issue: https://github.com/cyberbotics/webots/issues/2645 I adjusted a bit the soft CFM and left timestep at 16 and it looks better now. You can barely see it and only if you enable the bounding boxes view. Apart from that it doesn't  affect my work since it doesn't affect the physics but it looks like it is only a rendering issue.

ad-daniel, 2021-07-15:
I'm closing this for now, feel free to reopen it if you have additional issues.

closed 2021-07-15 by ad-daniel