cyberbotics/webots #2658: Simulation reset causes non-reproducible simulations

lukicdarkoo, 2021-01-15:
**Describe the Bug**
After the simulation reset, the behavior of the robots is not exactly the same. There are very tiny deviations, but enough to consider the simulations non-reproducible.

**Steps to Reproduce**
1. Download [reproducibility.zip](https://github.com/cyberbotics/webots/files/5821111/reproducibility.zip)
2. Reset the word a few times

ad-daniel, 2021-01-18:
I can't seem to replicate it, maybe I'm misunderstanding

- if you mean the box behaves differently after each call of `node.restartController()` in the supervisor then that's normal as only the controller is reset, not the physics. Calling `node.resetPhysics()` after `node.restartController()` gets the job done
[image]

- if instead you mean that manually resetting the world multiple times after letting it run a bit generates different CSVs, then I can't replicate on Ubuntu 20.04 (compiling from the latest master branch). PS: here I use your ZIP as is, without resetting the node physics
[image]
can't see much but here's the difference between the later CSVs and the first one
[image]

lukicdarkoo, 2021-01-18:
Try to restart the world manually using:
[image]

Then, compare two generated files using simple diff (instead of plotting).

lukicdarkoo, 2021-01-20:
@ad-daniel I updated the steps and the example, so it is easier to reproduce the "non-reproducibility". Sorry if you spend a lot of time because of my vague description.

lukicdarkoo, 2021-02-08:
I would leave this issue open until the second problem is resolved (described above, related to https://github.com/cyberbotics/webots/files/5856675/reproducability_proto.zip).