cyberbotics/webots #2661: Improve performance of Python RangeFinder getRangeImage

WasabiFan, 2021-01-18:
**Is your feature request related to a problem? Please describe.**

The `getRangeImage` and `getRangeImageArray` APIs both return Python `list`s. For my 640x480 image, the former takes ~10ms as measured by `time.time()` (i.e., real-world time) and the latter takes ~20ms. That time is, presumably, spent allocating a huge list (or list of lists). For comparison, the C `wb_range_finder_get_range_image` is functionally instantaneous, since it just returns a pointer to the existing buffer.

In my application, I don't need a `list`. A list is a silly data structure for an image 🙂If I'm shooting for a 60Hz controller, and want to run real-time, I have a 16ms tick budget. Spending 10ms retrieving the data and then another 10ms loading it into numpy is, in itself, more than 16ms.

**Describe the solution you'd like**

One option would be to return a `bytes` object like the Camera `getRangeImage` does. It may be a bit strange to expose the raw float layout in a language that otherwise wouldn't, but either way I'd be sufficiently happy with it since I could stuff it into `numpy`. Unfortunately, this would still require at least one copy to get it into a `bytes` and then another to load it into numpy for my use-case.

Another option would be to provide an interface returning a type which inherently shares the underlying buffer; the `bytearray` interface added for point clouds in #2321 would be great. It still suffers from the same slight weirdness as above, though.

A third option would be to directly expose `numpy` arrays as first-class data structures. It _feels_ like that may be worth doing for cameras/range-finders/etc. My _guess_ is that most users are loading the data into `numpy` either way if they're writing in Python, although I haven't done any research there.

**Describe alternatives you've considered**

Writing my controller in C or C++ could be done, but that's a silly reason to migrate. I could also potentially write just a shim for this portion in C and then invoke it via FFI. That's additional toolchain overhead which I'd rather avoid.

WasabiFan, 2021-01-18:
The above comparison was not exactly accurate: the original experimentation with the C API above was with a lower resolution. With the same 640x480 setting, in my testing the C API takes 1-3ms, typically, to grab a reading, with sporadic spikes to 20ms+. These spikes are enough to significantly increase the mean time to take a measurement. I also see these spikes in the Python API.

It looks like, if the frame isn't already at hand, there's some locking around the logic which updates some flavor of shared metadata. I don't see any obvious blocking calls so I'd guess the spikes to 20ms+ are due to some sort of lock contention -- I haven't looked into it though and I'm not sure if others would consider it an issue.

Source, for my own later reference: https://github.com/cyberbotics/webots/blob/master/src/controller/c/range_finder.c#L228-L249. I am running Windows. 

Meanwhile...

---

Brace yourselves for what you're about to read 🙂 

By ignoring the warnings and invoking the C API in the midst of a Python/C++ controller, I can get somewhat better performance...

```python
from controller import RangeFinder
import numpy as np

from ctypes import cdll, c_ushort, c_float, POINTER

webots_controller = cdll.LoadLibrary("Controller")
webots_controller.wb_range_finder_get_range_image.argtypes = [c_ushort]
webots_controller.wb_range_finder_get_range_image.restype = POINTER(c_float)

def range_finder_get_range_image(range_finder: RangeFinder) -> np.ndarray:
    width = range_finder.getWidth()
    height = range_finder.getHeight()

    tag = c_ushort(range_finder.getTag())
    ptr = webots_controller.wb_range_finder_get_range_image(tag)
    return np.ctypeslib.as_array(ptr, shape=(height, width))
```

The median is now 6ms (8ms mean) to get it into a usable numpy array. Better, but still worse than the plain C call -- I'm a bit surprised by the significant difference and not entirely willing to blame it purely on the FFI.

omichel, 2021-01-18:
Thank you for this nice investigation.
Please go ahead with the implementation of a new API function if you believe it's the way to go.
If it does provide a real performance improvement and is well designed, we will be happy to review your PR and merge it into Webots.

lukicdarkoo, 2021-01-18:
You can use the following PR for reference (the same problem we had with `getPointCloud`):
https://github.com/cyberbotics/webots/pull/2321/files

Like in the PR, I would leave the default behavior as is, and add a `data_type` parameter (which can be `list` or `buffer`). That way we preserve the backward compatibility.

WasabiFan, 2021-01-19:
Thanks for the thoughts! I will implement this and hopefully open a PR.

---

@lukicdarkoo The linked PR explains in the added documentation that:

> If `data_type` is equal to `buffer` then the function returns `bytearray` and it is fast as there is no memory copying.

It uses `PyBytes_FromStringAndSize`. Drawing from [the docs](https://docs.python.org/3/c-api/bytes.html#c.PyBytes_FromStringAndSize):

> Return a new bytes object with a copy of the string v as value and length len on success, and NULL on failure.

In other words, it seems like this `getPointCloud` API is returning a `bytes` rather than a `bytearray` and performing a copy in the process. This is the same behavior as the `Camera`. There is a [`PyByteArray_FromStringAndSize`](https://docs.python.org/3/c-api/bytearray.html#c.PyByteArray_FromStringAndSize) which would have the documented behavior, as I understand it.

Am I mis-understanding and/or was this intentional? It seems like the no-copy behavior would be ideal for performance, but _does_ have a different (unexpected) behavior if you retain the reference between sim ticks.

WasabiFan, 2021-01-19:
On the above:

I implemented both the `bytes` and `bytearray` variants and, evidently, the `bytes` version (which matches the current `getPointCloud` behavior and requires a full copy of the buffer) is _faster_ in my testing -- it takes about half the time. This is rather surprising and it isn't entirely clear to me why this is the case. Either way, since `bytearray` is mutable, it probably isn't a good fit for this.

WasabiFan, 2021-01-22:
The `data_type='buffer'` PR has been merged!

I figure this issue might be a good spot to consider design of e.g. a numpy interface or similar; does that seem reasonable?

omichel, 2021-01-22:
Yes, I believe this could be very useful indeed. Would you like to propose an implementation?

ygoumaz, 2022-11-08:
This issue is actually fixed. As this issue is quiet old, I will close it now. @WasabiFan, if you would like to go ahead with the numpy interface, I suggest you to open a new issue and PR.

closed 2022-11-08 by ygoumaz