cyberbotics/webots #2663: Increasing memory consumption over iterative reset simulation

angel-ayala, 2021-01-19:
I'm currently developing an reinforcement learning algorithm with Webots R2020b and recently has happened that the memory used by the webots-bin process reached out 5GB and more in some cases.
The scenario includes three robots Nodes and one custom proto Node, it is and ongoing project and still need some changes, I create and special scenario for replication called memory_check.wbt at in [the following repository](https://github.com/angel-ayala/gym-webots-fire)

**Steps to Reproduce**
1. Load the memory_check.wbt world.
2. Check the initial memory used by the webots-bin process
3. Run the simulation at fast as possible (CTL+4)
4. Iterative reset the simulation about 10 or 20 times
5. Check the memory increasing while the simulation is being reset.

As the problem it is continuously executed, it seems that the iterative reset simulation process does not leave enough time for the garbage collector algorithm to clear the memory or something like that, but after several times, as epochs needed in a RL problem the memory consumed by the simulator leaves with no memory to use by the algorithm freezing out and as consequently  ends the training process.

**System**
Different configuration system were used but.
 - Operating System: Manjaro (Arch based Linux distro), Linux kernel 5.10 x86_64
 - Graphics Card: AMD Radeon R5 M330 5GB.
 And
 - Operating System: Manjaro (Arch based Linux distro), Linux kernel 5.4 x86_64
 - Graphics Card: NVIDIA GeForce RTX 2060 6 GB.

Same results

angel-ayala, 2021-01-19:
It seems that the issue is still present in R2021a version

angel-ayala, 2021-01-20:
I checked and the issue is still present, I checked the memory maps through gnome monitor system and there is a [heap] identified item that is responsible of increasing the use of memory.

It is there any debug method to check how it is this happening in webots, maybe some compilation options?

Bests,

omichel, 2021-01-20:
This memory leak is probably related to some Webots node or API function. I believe the best way to localize it is to simplify your world file, step-by-step, by removing nodes or functionalities used by your controller program until the memory leak is not observed any more. This will allow you to identify the culprit: the node or API function that is causing the leak. Once you identified it, it will be way easier to track down the actual problem.

angel-ayala, 2021-01-21:
After checking each Robot node in the scenario, it appears to be a problem with my custom Proto Node, and its controller, the Node is very simple containing two display to show an sprite, one for fire and another for smoke.
The controller, change the image that must be showing each display.

The Proto Node is as follwing
```proto
# FireSmoke.proto
#VRML_SIM R2020a utf8
# license: Copyright Angel Ayala. Licensed for use with research purpose onlys.
# license url: https://www.github.com/angel-ayala
# tags: static
# Transparent images sequence to simulate a fire with smoke event.

PROTO FireSmoke [
  field SFString   name             "FireSmoke"         # Is `Solid.name`.
  field SFVec3f    translation      0 0.5 0             # Is `Transform.translation`.
  field SFRotation rotation         0 0 0 0             # Is `Transform.rotation`.
  field SFFloat    fireHeight       1                   # Is `Cylinder.height`.
  field SFFloat    fireRadius       0.2                 # Is `Cylinder.radius`.
  field SFString   controller       "fire_movement"     # Is `Robot.controller`.
  field MFString   controllerArgs   ""                  # Is `Robot.controllerArgs`.
  field SFString   customData       ""                  # Is `Robot.customData`.
  field SFBool     synchronization  TRUE                # Is `Robot.synchronization`
#  field SFBool     showSmoke    TRUE
]
{
%{ 
local smokeHeight = 25
if fields.fireHeight.value * 1.5 >= smokeHeight then 
  smokeHeight = fields.fireHeight.value * 1.5
end
}%
Robot {
  name IS name
  translation IS translation
  rotation IS rotation
  controller IS controller
  controllerArgs IS controllerArgs
  customData IS customData
  synchronization IS synchronization
  children [
    Display {
      translation 0 %{= smokeHeight * 0.3333}% 0
      children [
        DEF SMOKE_SHAPE Shape {
          appearance PBRAppearance {
            baseColorMap ImageTexture {
              repeatS FALSE
              repeatT FALSE
            }
            roughness 1
            metalness 0
          }
          geometry Cylinder {
            bottom FALSE
            height %{= smokeHeight }%
            radius %{= fields.fireRadius.value * 1.25}%
            top FALSE
            subdivision 8
          }
          castShadows FALSE
        }
      ]
      name "smokeDisplay"
      width 500
      height 320
    }
    Display {
      rotation 1 0 0 0
      children [
        DEF FIRE_SHAPE Shape {
[... 97 lines cut ...]
    def run(self):
        """Run controller's main loop.

        Change the sprite presented in the Display nodes.
        """
        # main control loop:
        print('Fire movement is active')
        counter = 1
        while self.step(self.timeStep) != -1:
            self.move_smoke(counter)
            self.move_fire(counter)
            counter += 1

    def __del__(self):
        self.smoke_screen.imageDelete(self.smoke_img)
        self.fire_screen.imageDelete(self.fire_img)

if __name__ == '__main__':
    # run controller
    controller = FireMovement()
    controller.run()

```

I noticed also, that the __del__ method is never reached, in any termination cases as mentioned [here](https://www.cyberbotics.com/doc/guide/controller-programming?tab-language=python#controller-termination)

omichel, 2021-01-21:
That if you force the deletion the image? E.g.;
```python
if __name__ == '__main__':
    # run controller
    controller = FireMovement()
    controller.run()
    del controller
```

angel-ayala, 2021-01-21:
Any line after controller.run() neither after the while loop are reacheble, I tested with some prints and nothing

omichel, 2021-01-21:
prints are not forwarded to the Webots console after this point, so it is not a proof.
Instead, you should try to save some dummy data into a file.
That should work.

angel-ayala, 2021-01-21:
Ok, I will test and check it out

angel-ayala, 2021-01-21:
Indeed, the code after those lines are being executed, however the issue is still present.
I double check if is that node, and it is, when I remove it from the scene leaving just the Mavic drone's Node the memory is stable, and when the FireSmoke Node is added, occurs this memory leak.

How can I proceed to solve this?

omichel, 2021-01-21:
OK, and if you simplify your controller (remove the code that loads sprites in the Display) do you still get the memory leak?

angel-ayala, 2021-01-21:
I did not executed the `load_sprites()` method and there is no memory leak.
It seems that the problem relies on the `Display.imageLoad()` method

omichel, 2021-01-22:
Yes, I believe the problem is now well isolated. It seems the [reset](https://github.com/cyberbotics/webots/blob/a05248b57540c72651f7353dfd34718c8ffaceba/src/webots/nodes/WbDisplay.cpp#L1200) method of the Display device is not called as it should be. Did you compile Webots from the sources? If so, you should be able to further investigate it.

angel-ayala, 2021-01-22:
I get it done, the `mImage` was not being reset properly

I leave the method as follow, and no memory leak no more
```c++
void WbDisplay::reset() {
  WbRenderingDevice::reset();

  delete[] mImage;
  mImage = NULL;
  mColor = 0xFFFFFF;
  mAlpha = 0xFF;
  mOpacity = 0xFF;
  mAntiAliasing = true;
  mUpdateRequired = true;
  mRequestImages = false;

  setup();
}
```

Today I can fork the repo, and pull the changes

commit 9f7fdb84 (2021-01-22): Fix #2663 memory leak on Display node (#2673)

* Fix #2663 memory leak on Display node

Display's image loaded in memory was not properly reset, when the simulation is reset iteratively, the Display node increased the memory use.

* Update changelog-r2021.md
closed 2021-01-22 by angel-ayala