cyberbotics/webots #2671: Errors in world file cause Webots to silently default to an empty world

WasabiFan, 2021-01-22:
**Describe the Bug**

When loading a world file, if there are errors (I've tried unmatched braces and incorrect property types such as swapping a number for a string), no error is printed and the window loads an empty (new) world.

**Steps to Reproduce**
1. Create a new world.
2. Insert some stuff. I added a chair and a light.
3. Save the world file.
4. Perform a fresh load of the world file. It loads fine.
5. Edit the world file and introduce an error. Originally, I discovered this with a string `"true"` instead of boolean `TRUE` in Robot `supervisor`. One similar example:
    ```
    PointLight {
      attenuation "foo"
      location 0 0.3 0
    }
    ```
    Alternately, removing a closing brace does it, or using a node type which doesn't exist.
6. Perform a fresh load of the world.
7. Observe that the world view is now black, the scene tree has reset to default, and there are no messages in the console. The window title still reports your file name.

**Expected behavior**

An error message in the console would be nice. Perhaps even stay on the existing world, or some other intermediate state. The completely empty scene view is confusing.

**Screenshots**

[image]

**System**
 - Operating System: Windows 10
 - Graphics Card: Machine 1: NVIDIA GTX 750 Ti, Machine 2: Intel Integrated Graphics

stefaniapedrazzi, 2021-02-10:
It works correctly on Webots start (i.e. when the failing world is passed as argument or it is the last open world) and the error messages are printed in the Webots console.
But it doesn't workd when loading the failing world from the Webots UI (Webots already started).

closed 2021-02-11 by stefaniapedrazzi

stefaniapedrazzi, 2021-02-11:
Fixed in #2737: the loading errors are now visible in the console.

> Perhaps even stay on the existing world, or some other intermediate state. The completely empty scene view is confusing.

I fixed the buggy behavior so that now it works exactly the same at Webots starts or when loading a world from the Webots application and it is possible to know where are the errors in the file.
But if you really think that will be better to handle differently the loading failure (stay on existing world or other) please feel free to open a new issue for this or a PR with your proposed solution.

PR #2737 (2021-02-11): Postpone sending logs to consoles during the world load
Fixes #2671:
logs generated during world load were sent to the consoles of the previous world and thus no longer visible once the load and the new world consoles were open.
This fix consists in queuing the logs during the world load and send them to the new console once they have been opened.