cyberbotics/webots #2682: https://github.com/cyberbotics/webots/archive/R2021a.zip could not work well

qingyi-wang, 2021-01-29:
click this link
https://github.com/cyberbotics/webots/archive/R2021a.zip
and use unzip to unzip. it doesn't work. download https://codeload.github.com/cyberbotics/webots/tar.gz/R2021a work well

qingyi-wang, 2021-01-29:
and tar -xvf have this error:
gzip: stdin: unexpected end of file
webots-R2021a/projects/default/worlds/textures/cubic/factory_back.hdr
tar: Unexpected EOF in archive
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now
work@work-Inc:~/Projects/03_nav$ ls
mrpt  mrpt-build-env  webots-R2021a  webots-R2021a.tar.gz

omichel, 2021-01-29:
I just tried to download the https://github.com/cyberbotics/webots/archive/R2021a.zip and it worked fine. I could also unzip it without any problem.
I also tried to download https://codeload.github.com/cyberbotics/webots/tar.gz/R2021a and it worked fine. I renamed it to R2021.tar.gz and uncompressed it with `tar xzf` successfully as well.
This is unfortunately not an issue we can address as it may be a bug with your local cache network configuration (cache or firewall?) or with GitHub?

closed 2021-01-29 by omichel

qingyi-wang, 2021-01-29:
OK.thanks, I'll try again.