cyberbotics/webots #2686: Something wrong about time stamp for ROS topic

JackyMao1999, 2021-01-31:
I'm using webots 2020b-rev1 now which I installed by .deb package. And I'm using ROS melodic + Ubuntu 18.04.
I have added the `-clock` and  `--use-sim-time` argument to the ros controller in webots.But I also notised that the time stemps are abnormal.

The /clock topic published by Webots like that:
[image]

When the sensors finished enabled ,I echo header of laser gpic, and I got:
[image]

I also echo another topic's header, I got:
[image]

When I run cartographer and move_base together. And I set `use-sim-time=true` in cartographer's launch file，like that：
[image]

The move_base wouldn't  run, because they have the different time stamp.
[image]

omichel, 2021-02-01:
Do you get the same problem with Webots R2021a?

JackyMao1999, 2021-02-01:
> Do you get the same problem with Webots R2021a?

I'm sorry.I haven't tried it.

JackyMao1999, 2021-02-02:
> Do you get the same problem with Webots R2021a?

I tried it with Webots R2021a, It has the same problems.

lukicdarkoo, 2021-02-02:
Do you use an external controller or the `ros` controller?

JackyMao1999, 2021-02-02:
> Do you use an external controller or the `ros` controller?

I use the ros controller，like this：

[image]

lukicdarkoo, 2021-02-02:
I can reproduce it, I am checking the problem

lukicdarkoo, 2021-02-02:
The `controllerArgs` parameter should contain also `--use-sim-time`:
```
  controllerArgs [
    "--name=robot_name"
    "--use-sim-time"
    "--clock"
  ]
```

And somewhere in the launch file you should set the `/use_sim_time` parameter to `true`:
```
<param name="/use_sim_time" value="true" />
```

Please let me know if this works for you.

JackyMao1999, 2021-02-03:
> The `controllerAr parameter should contain also `--use-sim-time`:
> ```vrml
> controllerArgs [
>   "--name=robot_name"
>   "--use-sim-time"
>   "--clock"
> ]
> ```
> 
> And somewhere in the launch file you should set the `/use_sim_time` parameter to `true`:
> ```xml
> <param name="/use_sim_time" value="true" />
> ```
> 
> Please let me know if this works for you.

I have tried this，but I still have this problem.

lukicdarkoo, 2021-02-03:
That is strange because it works fine for me. I modified this example:
https://github.com/cyberbotics/webots_ros/blob/master/launch/pioneer3at.launch

By applying the changes I mentioned above I could see the lidar sensor using the simulation time. Do you use Webots R2021a?

JackyMao1999, 2021-02-03:
> That is strange because it works fine for me. I modified this example:
> https://github.com/cyberbotics/webots_ros/blob/master/launch/pioneer3at.launch
> 
> By applying the changes I mentioned above I could see the lidar sensor using the simulation time. Can you try Webots R2021a?

oh,It works fine in Webots R2021a. I need to  change many code of the object to the new version.But thank you very much.

closed 2021-02-04 by lukicdarkoo