cyberbotics/webots #2694: Changing global coordinate frame in Webots

miroslavradojevic, 2021-02-02:
When simulating Lidar instrument, obtained 3D points (x,y,z) are having y value pointing upwards. Apparently, Webots global coordinate frame defines its y-axis to show upwards. 

I would like to have pointcloud values such that z-axis point upwards. There should be an option in the scene tree in WorldInfo where the directions of coordinate axes are selected.

Alternative was to change WorldInfo.gravity field to 0 0 -9.81, as stated in [here](https://stackoverflow.com/questions/54557696/changing-global-coordinate-frame-in-webots), but this additionally complicates model and introduces perks with setting other sensors. It just seems too complicated solution for a simple gereric feature.

There is a parameter of WorldInfo in Webots documentation, coordinateSystem that can take values ENU, NUE, or EUN. That looks like it should do the job. Unfortunately, it is not possible to find this field in the scene tree of Webots R2020a rev 1.

omichel, 2021-02-02:
I would recommend you to upgrade to Webots R2021a to have the WorldInfo.coordinateSystem parameter. As for the Lidar, should be able to orientate it with a 90° rotation along the X axis to get its Z-axis pointing upwards.

miroslavradojevic, 2021-02-02:
The problem with rotating Lidar object is that it has vertical field of view (verticalFieldOfView parameter) affiliated to y-axis, so if it is rotated then I have that angular constraint applied to horizontal direction and pointcloud becomes distorted. I will try with the Webots R2021a version, a general setting for axes directions.

miroslavradojevic, 2021-02-02:
I did try with R2021a however the effect is the same as if gravity vector would be changed in previous version, only that the setting is now separated in two components (gravity intensity and direction). Due to the gravity push objects move around the scene instead of standing in place. I am not sure now if the axis setting was actually doing what I expected it to do. Rotating Lidar would be a solution but then the vertical field of view (which is always narrower angular range than horizontal ones) occupies now a horizontal axis (since y is now horizontal and z vertical) and it is difficult to have adequate pointcloud values. If only there was a feature in Lidar that would specify which axis (N, E or U) is vertical, this would be solved.

miroslavradojevic, 2021-02-02:
It can be an argument to getPointCloud() function in Lidar API so that mapping into LidarPoint x,y,z values differs depending on the parameter.

correll, 2021-02-02:
I have the same problem. I'm working with Webots 2021a on MacOS and the y-axis points up, no matter which coordinate system representation I chose.

lukicdarkoo, 2021-02-02:
@miroslavradojevic I assume you use Python. Note that `getPointCloud` is very slow as it creates a list. Therefore, I strongly advise you to use `getPointCloud(data_type='buffer')` instead. See usage here:
https://github.com/cyberbotics/webots_ros2/blob/8230dd586462c4754f5eb4fd2022f8428ff94a90/webots_ros2_core/webots_ros2_core/devices/lidar_device.py#L96

Regarding the axis order, why is it important to you? You simply swap the indices when accessing the points. Or, you can swap the columns once the numpy array is created and reshaped.

lukicdarkoo, 2021-02-02:
@correll What is the orientation of the arena? I have feeling that you should rotate the arena. 
By selecting `Top View` (https://cyberbotics.com/doc/guide/the-user-interface#view-menu) you should see the arena from above. If it is not the case then the arena is not properly oriented.

correll, 2021-02-02:
You are correct! It was not clear to me that changing the coordinate system also requires rotating all the objects. This also explains why the robot just took off north (it fell down). Thanks for your quick reply.

lukicdarkoo, 2021-02-03:
> You are correct! It was not clear to me that changing the coordinate system also requires rotating all the objects. This also explains why the robot just took off north (it fell down). Thanks for your quick reply.

The `coordinateSystem` property doesn't affect the coordinate system of the PROTOs, so you have to rotate them manually. However, we are working on that.

If you need to rotate a lot of objects in the world (e.g. from NUE to ENU) you can use a converter:
https://github.com/cyberbotics/webots/tree/master/scripts/converter

miroslavradojevic, 2021-02-03:
> @miroslavradojevic I assume you use Python. Note that `getPointCloud` is very slow as it creates a list. Therefore, I strongly advise you to use `getPointCloud(data_type='buffer')` instead. See usage here:
I use c++, if I use python I will use buffer option. In c++, there is an API function for obtainin pointcloud named 
```
const LidarPoint *getPointCloud() const;
```
Which maps points into LidarPoint struct and the vertical axis coordinate is mapped into y field, instead of z. 

https://github.com/cyberbotics/webots_ros2/blob/8230dd586462c4754f5eb4fd2022f8428ff94a90/webots_ros2_core/webots_ros2_core/devices/lidar_device.py#L96
> 
> Regarding the axis order, why is it important to you? You simply swap the indices when accessing the points. Or, you can swap the columns once the numpy array is created and reshaped.

Real lidar and lidar pointcloud processing methods expect vertical axis mapped to z of the point. If I switch from Webots lidar values to real lidar readouts there is a need to adjust pointcloud processing code, or do  the y-z swapping. It is not a problem per se (I did swap those once read), just that it is either additional computation or code that works with lidar needs to be refactored for webots output. Instead I had impression that a single argument in getPointCloud() would solve this.

miroslavradojevic, 2021-02-03:
> I have the same problem. I'm working with Webots 2021a on MacOS and the y-axis points up, no matter which coordinate system representation I chose. See below. My understanding is that changing to ENU should make the z-axis point up.

It is true that ENU makes z-axis point up but from what I understood the gravity still pushes along y-axis, which causes objects to move sidewise (north). So the whole scene/objects would need to be rotated to adjust to the new gravity direction. Then the viewpoint also needs to take into account to the rotation.

omichel, 2021-02-08:
I would recommend you to check the [enu_coordinate_system.wbt](https://github.com/cyberbotics/webots/blob/master/projects/samples/howto/enu_coordinate_system/worlds/enu_coordinate_system.wbt) in Webots R2021a which is a good example for using the ENU coordinate system in a Webots simulation.

closed 2021-11-15 by BenjaminDeleze