cyberbotics/webots #2722: 3d model texture path issue

mxf8bv, 2021-02-08:
When importing 3d models (e.g. .obj with .mtl files), textures are often not found.
On Windows it seems to work to have them in the same folder as the model only when the path is prefixed like this: ".\texturename.jpg" (but not "./texturename.jpg" in Unix style URIs, which again makes it less portable).

On Linux, I could not get relative paths to workat all and textures only seem load when absolute paths are given (which is obviously not portable).

Furthermore, the model geometry seems to be saved in a world file but textures are not, so that when models move in the file system, a gray untextured render remains.

Tested with Linux and Windows in Release R2021a but the same effect also occured in earlier releases.

omichel, 2021-02-08:
Could you please post a simple example showing this problem: a ZIP file including a world file, a obj file and a texture?

BenjaminDeleze, 2021-11-15:
I'm closing this for now, feel free to reopen it if you have additional issues.

closed 2021-11-15 by BenjaminDeleze

loicgirard, 2022-03-09:
Personnaly, I have this Boat.obj mesh, with its corresponding Boat.mtl . The mtl file points to .bmp files in the textures folder.

In WeBots I can only import the Boat.obj model using import 3D model or using 
`
geometry Mesh {
  
`
which appears white. So I don't understand how to add the .mtl so that the boat gets its textures. 

Here's the zip file.
[Fishing_Boat_obj.zip](https://github.com/cyberbotics/webots/files/8212987/Fishing_Boat_obj.zip)

omichel, 2022-03-09:
The .mtl file is ignored by Webots indeed. Here is how to proceed:

1. You should convert your images to JPG format.
2. You should refer to these images in your [PBRAppearance](https://cyberbotics.com/doc/reference/pbrappearance) node.

See for example: [CorrugatedMetal.proto](https://github.com/cyberbotics/webots/blob/released/projects/appearances/protos/CorrugatedMetal.proto).

omichel, 2022-03-09:
I believe the mapping is the following:
- occlusionMap: boat_ao.jpg
- normalMap: boat_normal.jpg
- baseColorMap: boat_diffuse.jpg
- roughnessMap: boat_gloss.jpg
- metalnessMap: boat_specular.jpg

loicgirard, 2022-03-09:
I was also wondering how I could add the other textures, thanks! I was using Appearance to add boat_diffuse.jpg to the Mesh node, but the texture isn't matching, I explain:

In my case the textures aren't uniform and look like this: 

[image]

When I add the boat_diffuse.jpg to the baseColorMap, I get the following result:

[image]

I think I should change the position and orientation of the texture, I tried to move it with the TextureTransform node of the Appearance node, but I haven't found a match yet :/

omichel, 2022-03-09:
Normally, the obj file should contain the texture mapping (texture coordinates). Can you open it successfully in a 3D modeling software like Blender?

omichel, 2022-03-09:
I just tried and it works nicely in Webots for me:
[image]

loicgirard, 2022-03-14:
Thanks omichel !

I am using 2021b, and I still have the texture not well put with your worlds.zip :( I guess it works only for >2022a-b

omichel, 2022-03-14:
> I guess it works only on >2022a-b

Yes, I would recommend you to upgrade to Webots R2022a.