cyberbotics/webots #2730: All of the 3D lidar sensor give back wrong Z-axis data by webots_ros

RENyunfan, 2021-02-10:
**Describe the Bug**

All of the 3D lidar sensors give back the wrong Z-axis data by webots_ros.

**Steps to Reproduce**
1. Create a new world and set some obstacles.
2. Creat a robot and add a 3d lidar sensor such as velodyneVLP-16.
3. Select the contoller as ros.
4. Use webots_ros to publish the point cloud.
5. Move the lidar to 1m (or 2m) above the ground.
6. See in rviz, the ground is -1.3 m ( -2.54m).

[image]

[image]

And I also test sick and VLP-64, the same thing happened, which is unreasonable.

I didn't write any code, the things just went wrong with ros service.

Any idea?

RENyunfan, 2021-02-12:
I tested the point cloud in the x, y-axis, both of them are right, and I also have some z-axis sample:
```matlab

y = [-1.29139757156,-2.55461740494,-3.80381822586,-5.05427646637];
x = 1:4;
plot(x,-y,'*-')
```
and finally, find the scaler is close to 1.26-1.27.
[image]

Anyone can give some explanations?

Simon-Steinmann, 2021-03-09:
@RENyunfan can you share your worldfile? That would make it much easier to troubleshoot.

closed 2021-11-15 by BenjaminDeleze