cyberbotics/webots #2736: Lidar distances are somtimes too big if the ray hits an edge

Thomas-Feldmeier, 2021-02-10:
**Describe the Bug**
the lidar rays at the left/right edge of an obstacle sometimes measure a too big value, generating false data (see picture).
there are some blue points in empty space.

**Expected behavior**
It should be possible to use lidars without this problem. Maybe a parameter? (real lidars have less errors)

*Screenshots*

[image]

(implementation: the z-buffer distance of only one pixel at the center of the ray instead of averaging a lot of pixels?)

lukicdarkoo, 2021-03-02:
@mael25 There is a chance it is related to #37, maybe related to the spherical deformation

lukicdarkoo, 2021-05-06:
I have a theory for this one.

The point cloud is calculated from a depth camera that has the same horizontal resolution as the lidar:
https://github.com/cyberbotics/webots/blob/60d017e8f48caf1863f5d3a93f3937d3bc682075/src/webots/nodes/WbAbstractCamera.cpp#L497

Let's say we have a high-resolution depth camera that detects an edge of an object as follows:
`| wall | wall | wall | object | object | object |`

Now, try to reduce the resolution of the camera by x2:
`| wall | wall/object | object |`
Instead of encoding the edge in two pixels, it is now encoded in a single pixel. Which means that the edge has to be averaged.

To avoid the problem we can take high-resolution cameras and pick every nth pixel. But it is very expensive.

Another solution would be to tell OpenGL the way the edge pixels are encoded. I didn't check the part of the code that generates depth images, but I guess it is possible.

mael25, 2021-06-23:
I have some precisions on the issue after investigating:

- It occurs only if the `spherical` field of the `Lidar` is set to `true`.
- In this case, one or multiple sub-cameras are activated according to the needs (verical/horizontal `fieldOfView`). Their resolution is different than the one set for the `Lidar` but the output of the `merge_spherical.frag` shader meets the resolution of the `Lidar`.
- The bug still appears if we have a `fieldOfView` requiring only one sub camera and force it to have the same resolution as the `Lidar`. In this case, the setup of the camera is the same whatever the value of `spherical` is. The only change is that in the `spherical=TRUE` case, it goes through the `merge_spherical.frag` shader.
- If we keep the case where the setup is similar for both and remove the rectification of the spherical transform at the end of `merge_spherical.frag` the issue still occurs.
- But if we change how the depth is computed: `texture(inputTextures[0], texUv)` instead of `texture(inputTextures[0], faceCoord)` then we get the exact same result and there is no issue anymore.
- **That leads to think the issue is probably related to the way these `faceCoords` are computed.**
- The bug appears not only for the `Lidar` but also for the `RangeFinder`.

closed 2021-07-09 by mael25