cyberbotics/webots #2744: Incorrect handling of greek characters in file path cause Webots crash when starting the program from a wbt file

it14149, 2021-02-11:
**Describe the Bug**
If you try to start Webots from a wbt file (double-click) that is located inside a folder with a greek name, Webots will start, show a "Couldn't open file" error and instantly terminate. The file path shown has gibberish letters in the greek folder part of it
[image]

Opening the file from the open world menu works fine, but this issue may happen with other characters as well

**Steps to Reproduce**
1. Create a folder with greek characters
2. Place the 'worlds' folder with a wbt file inside
3. Double click on the wbt file

**Expected behavior**
Webots should normally load the world

**Screenshots**
https://user-images.githubusercontent.com/12613442/107705062-deaaec00-6cc6-11eb-9841-ad28f7e232ad.png

**System**
 - Operating System: Windows 10 Pro
 - Graphics Card: NVIDIA GeForce GTX 1050 6GB

omichel, 2021-02-12:
I believe I fixed this bug. Can you download, unzip and copy this [webotsw.zip](https://github.com/cyberbotics/webots/files/5973332/webotsw.zip) file in `C:\Program Files\Webots\msys64\mingw64\bin\webotsw.exe`, overriding the existing one and let me know if that fixes the problem for you?

omichel, 2021-02-15:
This bug should be fixed in the upcoming nightly builds, starting from tonight.

PR #2750 (2021-02-15): Fixed handling of UTF-8 paths for world files opened from Windows GUI
Fixes  #2744.
closed 2021-02-15 by omichel