cyberbotics/webots #2758: Webots crashes when run the samples

tyc877, 2021-02-17:
**Describe the Bug**
Crashes happen when running the samples, the error code is 0xc0000279

**Steps to Reproduce**
1. Go to Webots
2. Click on samples
3. Run the samples
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Operating System: Windows 10 20H2
 - Graphics Card: NVIDIA GeForce GTX 1650Ti

**Additional context**
Add any other context about the problem here.

lukicdarkoo, 2021-02-17:
You cannot run any sample on Windows?

tyc877, 2021-02-17:
no, I cannot
0xc0000279 error happens

tyc877, 2021-02-17:
The cpu is amd Ryzen7 4800H

lukicdarkoo, 2021-02-17:
Can you turn off your antivirus software?

tyc877, 2021-02-17:
There is no antivirus software except windows defender.

tyc877, 2021-02-17:
And in the console, the messages are like "The process crashed some time after starting successfully" and "controller crashed".

omichel, 2021-02-17:
OK, so it is not Webots that is crashing, but the controller program, right?

tyc877, 2021-02-17:
you are right, it happens once I run the sample with the controller.

tyc877, 2021-02-17:
May I know how to fix this problem?

omichel, 2021-02-17:
It could be a DLL conflict: the controller which is launched by Webots loads a number of DLLs from Webots and from your system. If you have installed some third party software that put some DLL in the wrong way (like in the Windows system folder, which is bad), then, it may cause the Webots controllers program to crash. Did you install any special software on your PC?

tyc877, 2021-02-17:
I installed Matlab, SolidWorks, and CoppeliaSim.
I guess I did not change any DLL files manually.
How can I check this DLLs problem and find which DLL is conflicted?

omichel, 2021-02-17:
You can investigate with [dependency walker](https://www.dependencywalker.com/) on the binary file of a Webots controller that is crashing. It should list all the DLL used by the controller. They should be either Windows system DLL or Webots DLLs. If a a DLL is coming from another software, this is likely the culprit.

tyc877, 2021-02-17:
Ok, I will try, thank you very much.

omichel, 2021-02-17:
You should instruct dependency walker to search also in the Webots DLLs, i.e., in the `C:\Program Files\Webots\msys64\mingw64\bin\`, `C:\Program Files\Webots\msys64\mingw64\bin\cpp\` and `C:\Program Files\Webots\lib\controller\` folders.

tyc877, 2021-02-17:
Ok, thank you very much.

ad-daniel, 2021-07-15:
I'm closing this for now, feel free to reopen it if you have additional issues.

closed 2021-07-15 by ad-daniel

tyc877, 2022-03-28:
Thank you very much!