cyberbotics/webots #2764: 'wb_supervisor_node_get_velocity' returns 3 elements in Matlab

Suke0811, 2021-02-18:
**Describe the Bug**
The [document ](https://www.cyberbotics.com/doc/reference/supervisor?tab-language=matlab#wb_supervisor_node_get_velocity) says that 'wb_supervisor_node_get_velocity' returns 6 elements, [3 translation velocity, 3 angular velocity]. However, in Matlab this only returns 3 elements. 

The [Matlab API](https://github.com/cyberbotics/webots/blob/master/lib/controller/matlab/wb_supervisor_node_get_velocity.m) indeed returning 1 by 3 elements.
Other languages seemingly return 6 elements.

*Version*
Webots R2021a

stefaniapedrazzi, 2021-02-18:
It should be fixed in #2765.
@Suke0811 you should also be to test the fix simply by modifying the file in your local installation.
If you will do it, please let us know if the issue is gone.

Suke0811, 2021-02-18:
Thank you for the response.
I changed the element size to 6, and I got 3 angular velocities now. The issue is gone.

- setdatatype(obj,'doublePtr', 1, 3); => setdatatype(obj,'doublePtr', 1, 6);

Hope this fix comes out soon for everyone.

closed 2021-02-18 by Suke0811

stefaniapedrazzi, 2021-02-18:
Thank you for the feedback!

The fix will be available in the nightly build release of R2021-rev1 tomorrow
https://github.com/cyberbotics/webots/tags

PR #2765 (2021-02-18): Fix vector size
Fix #2764: fix the wrong size when converting the C vector to Matlab.