cyberbotics/webots #2767: Barometer/Altimeter

lukicdarkoo, 2021-02-19:
It would be nice to have an altimeter to measure altitude. At the moment, we can measure the altitude with GPS, but the altimeter would be a better formalization of the altitude measurement.

sishamilton, 2021-02-26:
Hello, I am interested in working on this issue. Do you have any advice as to where to start? Thanks for the help.

omichel, 2021-02-27:
1. Install the [Webots development environment](https://github.com/cyberbotics/webots/wiki) and compile Webots from the source.
2. Follow these [instructions](https://github.com/cyberbotics/webots/wiki/Adding-New-Node-and-API-Function) to add a new node.
3. The `Altimeter` node should be strongly inspired from the [GPS](https://github.com/cyberbotics/webots/blob/master/src/webots/nodes/WbGps.cpp) node (you can use it as a template to get started).
4. Post any question here, we will be happy to help you.

sishamilton, 2021-04-01:
Does it make sense to retain the noise correlation used by the GPS node in the Altimeter, or should I only include simple Gaussian noise?

omichel, 2021-04-01:
To me the noise correlation doesn't seem relevant in the case of the altimeter.
So I would suggest to use a simple Gaussian node instead.
However, I am not an expert of altimeters.
If you have some ideas to make the measurements more realistic with respect to real altimeters, feel free to propose any new parameter.

stefaniapedrazzi, 2021-06-30:
Added in  #3241

closed 2021-06-30 by stefaniapedrazzi