cyberbotics/webots #2770: Jerking motions move touch sensors

timonegk, 2021-02-19:
**Describe the Bug**
When a robot's joints are moved suddenly in a jerking motion in webots, touch sensors get delocated (see video). When I change the TouchSensors in the proto file to Solid, the issue does not occur.

**Steps to Reproduce**
1. Use [this world](https://github.com/bit-bots/wolfgang_robot/blob/50873a3f8d365d21713cca2d838af5910d70086b/wolfgang_webots_sim/worlds/flat_world.wbt) including [this proto file](https://github.com/bit-bots/wolfgang_robot/blob/50873a3f8d365d21713cca2d838af5910d70086b/wolfgang_webots_sim/protos/Wolfgang.proto)
2. Start webots
3. Run this python script:
```
from controller import Supervisor

supervisor = Supervisor()
robot_node = supervisor.getFromDef("Robot")

for i in range(100):
    supervisor.step(1)
    robot_node.resetPhysics()
    supervisor.getDevice('RAnklePitch').setPosition(i % 2)
```

**Expected behavior**
The touch sensors should stay where they are.

**Screenshots**

https://user-images.githubusercontent.com/25013222/108554537-80988d00-72f4-11eb-9c32-27c0e14662f9.mp4

**System**
 - Operating System: Debian Testing, Ubuntu 18.04
 - Graphics Card: NVIDIA GeForce RTX 2080 Ti

timonegk, 2021-05-06:
Hi, I just noticed that this also happens without TouchSensors and without `resetPhysics`. It looks like this: 
[image]
I am using [this proto file](https://github.com/bit-bots/wolfgang_robot/blob/27327f0c90d72613aa040fee932956ecd58ef2c5/wolfgang_webots_sim/protos/Wolfgang.proto), and the problem also occurs when each HingeJointWithBacklash is replaced with HingeJoint ([proto file](https://github.com/bit-bots/wolfgang_robot/blob/6a799206a10e6b7497b63d63c0d89aaf2899707d/wolfgang_webots_sim/protos/Wolfgang.proto)).
Minimal reproducible code for just one motor is the following:
```python
import random
from controller import Supervisor

s = Supervisor()
tf = s.getFromDef('amy').getField('translation')
motor = s.getDevice('RElbow')

for _ in range(1000):
    tf.setSFVec3f([0,0,0.43])
    motor.setPosition(random.uniform(motor.getMinPosition(), motor.getMaxPosition()))
    s.step(8)
```

lukicdarkoo, 2021-06-25:
@timonegk I cannot reproduce this bug in R2021b anymore, maybe it got fixed as a part of the other fixes