cyberbotics/webots #2779: Toggle rendering from supervisor API

timonegk, 2021-02-24:
**Is your feature request related to a problem? Please describe.**
In #2286, a separate rendering button that toggles the rendering has been introduced to be used instead of the previous fast mode. I am starting webots from python and would like to disable the rendering using the supervisor API.

**Describe the solution you'd like**
Similar to `wb_supervisor_simulation_set_mode` there could be `wb_supervisor_simulation_disable_rendering` to disable the rendering.

**Describe alternatives you've considered**
* Manually disabling the rendering in the opened webots window is possible but cumbersome
* Starting webots with `--no-rendering` as command line argument would be possible but cannot be used to toggle the rendering.

omichel, 2021-02-24:
I don't understand why the `--no-rendering` command line option is not suitable for your use case. Do you need to have rendering enabled at the beginning and then after some time disable it from the supervisor?

timonegk, 2021-02-24:
No, you are right. I could use the `--no-rendering` flag. Currently the function in my code where webots is started doesn't know about the mode it will be running in but I could change that.
I had the feeling that the option was not added to the supervisor by accident, not by choice, though. The simulation mode can also be set from both the command line argument and the supervisor API; and all of the other buttons (reset, step, pause, fast, and the recording features) are available from the API, so I don't see why disabling the rendering should not be possible.

omichel, 2021-02-24:
> I had the feeling that the option was not added to the supervisor by accident, not by choice, though. The simulation mode can also be set from both the command line argument and the supervisor API; [...]

Simply because there is no use case for this so far.

lukicdarkoo, 2021-02-24:
If needed we can add it?

timonegk, 2021-02-24:
No problem, I'm going to close this and will reopen if there is a use case.

closed 2021-02-24 by timonegk