cyberbotics/webots #2788: Typo in the documentation of Supervisor Programming code example of C++

bkpcoding, 2021-02-28:
**Documentation page(s) concerned**
https://github.com/cyberbotics/webots/blob/4a35ce8669f933775f70824dd7f37cac9d30a51e/docs/guide/supervisor-programming.md

**What is the problem with the current documentation?**
The condition for the "if" loop mentions "node_robot" while the actual object is named "robot_node", which I think is a simple typo.

[image]

**Describe how you would fix it**
Just want to clear the typo by fixing "node_robot" to "robot_node", and will create a pull request for the same in a while. Please, correct me if I am wrong somewhere.

omichel, 2021-02-28:
Yes, you are right. Please go ahead with the PR. Thank you!

bkpcoding, 2021-02-28:
Sure, I have submitted PR you can check it and I am currently going through documentation and setting up the development environment, will work on other issues soon.

closed 2021-02-28 by bkpcoding