cyberbotics/webots #2791: display a predicted trajectory

Technician13, 2021-03-01:
My webots's version is 2020b, now I have built a quadruped robot in webots and I have planned a swing trajectory for its foot, what should I do to show this predicted trajectory in the simulation interface?

lukicdarkoo, 2021-03-01:
You can display it in the robot window:
https://cyberbotics.com/doc/guide/controller-plugin?version=master#robot-window

Something similar to what this guy did:
https://www.youtube.com/watch?v=fISDutZca5g

stefaniapedrazzi, 2021-03-01:
Here is a sample simulation where the path is drawn directly in the Webots 3D scene:
https://www.cyberbotics.com/doc/guide/samples-howto#supervisor_draw_trail-wbt

Technician13, 2021-03-01:
> You can display it in the robot window:
> https://cyberbotics.com/doc/guide/controller-plugin?version=master#robot-window
> 
> Something similar to what this guy did:
> https://www.youtube.com/watch?v=fISDutZca5g

Thank you for your reply,waht I mean is that the predicted trajectory displayed in webots 3D scene window,I think this is more intuitivei.

Technician13, 2021-03-01:
> Here is a sample simulation where the path is drawn directly in the Webots 3D scene:
> https://www.cyberbotics.com/doc/guide/samples-howto#supervisor_draw_trail-wbt

Thank you for your reply, this result is similar to the Pen node,the trajectory in this result is the path that robot has finished,what I mean is display the trajectoy that is planned by my algorithm and not been finished by robot.

stefaniapedrazzi, 2021-03-01:
> Thank you for your reply, this result is similar to the Pen node,the trajectory in this result is the path that robot has finished,what I mean is display the trajectoy that is planned by my algorithm and not been finished by robot.

Yes. This is just an example but you can adapt it to draw the path all at once.
I would suggest you to look at the sample controller, try to understand it and modify it to your needs.

Technician13, 2021-03-01:
> Yes. This is just an example but you can adapt it to draw the path all at once.
> I would suggest you to look at the sample controller, try to understand it and modify it to your needs.

Okay,thank you very much

closed 2021-03-01 by stefaniapedrazzi

HimGautam, 2021-03-05:
How can I add pen to Epuck robot to plot the trajectory while moving.

Technician13, 2021-03-06:
> How can I add pen to Epuck robot to plot the trajectory while moving.

https://www.cyberbotics.com/doc/guide/samples-howto#supervisor_draw_trail-wbt
you can look at this project, I think it is helpful for your need, or you can browse the help document and there will be detailed introduction about "Pen", however, I still think this project can meet your needs.