cyberbotics/webots #2792: Minor change in webots reference doc of Node chart

bkpcoding, 2021-03-01:
**Documentation page(s) concerned**
https://cyberbotics.com/doc/reference/node-chart#inheritance

**What is the problem with the current documentation?**
I think that the documentation needs to state that the arrow head points to the derived node while the arrow tail starts with the base node. 
[image]

**Describe how you would fix it**
Again correct me if I am wrong. If the issue is fine then I think it's a good first issue for the new contributors.  They can just change the arrow head and tail part of the documentation by clicking the link above the documentation "Contribute on Github!".

stefaniapedrazzi, 2021-03-01:
Yes, this is wrong and should be fixed as soon as possible.
But I would not wait for new contributors for such a minor change and I would fix it immediately.
If you won't do it yourself, we will do it.

bkpcoding, 2021-03-01:
Oh okay, sure I will fix it immediately.

closed 2021-03-01 by bkpcoding