cyberbotics/webots #2798: 4k display UI text is too small

blueshrapnel, 2021-03-01:
**Describe the Bug**
All text, except for the source in the editor is illegible.  The video icons are also tiny, but I can live with that.

**Steps to Reproduce**
open Webots on a 4k monitor.

**Expected behavior**
Tiny font.

**System**
 - Operating System: Linux Ubuntu 20.04

**Additional context**
Simple fix that worked on my computer.

Suggetion: change the style sheet `$(WEBOTS_HOME)/resources/stylesheet.linux.qss`, replace the `px` with `pt`.  pixel sizes don't scale, points do.  Most people have a scale setting in their display settings, which this works with.  I also dropped all the sizes by 1 pt and that suited me.

The icons are still too small.

omichel, 2021-03-01:
It should be fixed in #2631. Can you try a nightly build and confirm it works for you?

blueshrapnel, 2021-03-01:
Perfect, thank you it is.  Icons are clear  and text legible now.

closed 2021-03-01 by blueshrapnel