cyberbotics/webots #2826: Add flag to prevent changing channel for Receiver and Emitter

Adman, 2021-03-08:
**Is your feature request related to a problem? Please describe.**
We are working on a simulator where 2 teams (each having 3 robots) are playing soccer - [link](https://github.com/RoboCupJuniorTC/rcj-soccer-sim). We would like to enable communication between them via `Receiver` and `Emitter` nodes -- each team having different channel to communicate through. However, it is now possible to change the channel via `setChannel` function and send "unexpected" message, or spam the other team with a bunch of messages.

**Describe the solution you'd like**
It would be great if we could specify something like `cannot_change_channel true` flag in the world file for `Emitter` and `Receiver` nodes.

**Describe alternatives you've considered**
I haven't found any other alternative how to make the communication between robots available.

**Additional context**

Simon-Steinmann, 2021-03-09:
You could have your Supervisor / referee controller asign a random channel to each team at the start. That way they dont know the channel of the other team. Or you just look through the submitted controllers and make sure they dont contain any supervisor functionality and things like `setChannel`

Adman, 2021-03-09:
@Simon-Steinmann It's not that easy to just search for `setChannel` because you can easily "hack" it like this

```python
>>> class A:
...     def setChannel(self, value):
...             print("called setChannel")
... 
>>> a = A()
>>> x1 = "set"
>>> x2 = "Channel"
>>> getattr(a, x1 + x2)(5)
called setChannel
```

Maybe assigning a random channel would help a little, but still doesn't solve the issue, because you can iterate over the channels, sniff for communication and then you know the opponent's channel number.

@omichel Thanks, this is actually the best possible solution. When do you plan to release new Webots version? RoboCupJunior competition will be held in June - we need to have code frozen during April, so I'm just wondering if we can make it till that, otherwise we will either stick with random channels or make the communication not possible.

closed 2021-03-18 by Adman

Adman, 2021-03-19:
@omichel Hello. I just downloaded nightly build `Webots Nightly Build (18-3-2021)`, but I'm getting `Skipped unknown 'allowedChannels' field in Emitter node.` error. It is defined like this

```
    Emitter {
      range 10
      baudRate 115200
      channel 2
      allowedChannels [2]
      name "team emitter"
    }
```

Do you have any idea what could go wrong?

omichel, 2021-03-22:
Yes, it should be included in the `R2021a-rev1` nightly build now, but it will be integrated in the `R2021b` build only after we merge the master branch into the develop branch which happens from time-to-time, but not necessarily every day, so be patient, it will make its way in `R2021b` as well soon.

commit fc6a9b30 (2021-03-18): Ability to restrict Emitter/Receiver channels (#2849)

* Ability to restrict Emitter/Receiver channels

* closes #2826

* Update changelog-r2021.md

* Update changelog-r2021.md

* Update docs styles

* Reset channel to the first allowed one when editing

* update allowed channels when editing from GUI

* realloc allowed channels array when editing

* reset channels in the gui if they are not allowed

* English style

* English style