cyberbotics/webots #2828: Usually the PROTO nodes we provide are by default added above the ground.

omichel, 2021-03-09:
Usually the PROTO nodes we provide are by default added above the ground.
With this change the ball will be half below the ground.
Is this really wanted? Because version seems better and this is the only ball from the `objects/balls` section that is not located above the default ground position.

_Originally posted by @stefaniapedrazzi in https://github.com/cyberbotics/webots/pull/2783#discussion_r589554149_

Simon-Steinmann, 2021-03-11:
I think 0 0 0 is the most intuitive. Adding a new Object or solid, will place it at the origin of the parent. I think this consistency outweighs the "nice to have" feature of spawning an object at the "correct" height. If something like this would be added, I think it would require some LUA script to place it according to the coord-sys. And perhaps only, if it is a parent of root.

omichel, 2021-05-26:
I believe this issue has been fully addressed in #2865.

PR #2865 (2021-03-17): Fixed balls protos with orientation and position rules
Fixes #2828.

This PR fixes the default center and orientation for ball protos.
In addition, it fixes the icon for the RobocupSoccerBall proto which was missing from the doc and not nice from the proto icon.
The object doc was regenerated from the script, see https://cyberbotics.com/doc/guide/object-balls?version=fix-ball-protos-with-orientation-and-position-rules
closed 2021-05-26 by omichel