cyberbotics/webots #2847: Realtime factor is not steady

YannickdeHoop, 2021-03-11:
I try to simulate 25 lidar sensors on 34hz with a realtime factor of 1.00. During the simulation the realtime factor is jumping between 0.7 en 1.0. The GPU and CPU usage doesn't seem really high. 
When I set the simulation on 'fast' the realtime factor will increase and will jump between 1.95 and 2.05. So this seems to be far more steady than simulating with a realtime factor of 1.00. 
I am wondering what could possibly be the cause of this 'jumpy' behavior when simulating with a realtime factor of 1.00?
And are there parameters that we could adjust to get a steady realtime factor?

Simon-Steinmann, 2021-03-11:
Can you share your project files? 25 lidar sensors will produce lots of data. The bottleneck might be the memory buffer, leading to inconsistent times to calculate each step, which makes it hard to keep it steady at 1. Profiling your controller execution could also give more insight.

YannickdeHoop, 2021-03-12:
@Simon-Steinmann thanks for your quick response. We use a custom controller plugin in combination with ROS. I made a more simple example. In this example you see also very jumpy behavior of the realtime factor. 

[webots_project.zip](https://github.com/cyberbotics/webots/files/6128686/webots_project.zip)

Simon-Steinmann, 2021-03-12:
First of all, I would set the interval to 30ms instead of 29   (34hz would be 29.4). This way you always get a proper value when using a step size of 10ms. 
Secondly, you will never get a perfect realtime factor. That is why you should publish the simulation time `robot.getTime()` as your rostime. Then it does not matter how fast the simulation runs. And your controllers need to be actually running. I dont know what you do in your actual project, but what you send me, the controller just enable the lidar and then exit.

YannickdeHoop, 2021-03-15:
Thanks, we will take a look at the interval of the sensor.
We are currently using the `robot.getTime()` as rostime. But I use real hardware in combination with the simulation. I have a couple of separate computers that is running the robot software. The simulation provides the sensor data to these computers. So that's why it is necessary to have a stable realtime factor. In that case it doesn't really matter if the realtime factor jitters a little bit around 1.0. When the realtime factor jitters to much the "robots" will give errors on the rates of certain topics.
yes, I did send you a simple example without a running controller because in that case I saw also the jittering realtime factor. I would expect that the realtime factor should also be stable without a running controller?

In the source code I found this: https://github.com/cyberbotics/webots/blob/master/src/webots/engine/WbSimulationWorld.cpp#L150
Do you think we could gain some stability here? Now there are some empirically found numbers ... 
Do you have other suggestions were we could gain some stability in real time factor?

lukicdarkoo, 2021-03-15:
> In the source code I found this: https://github.com/cyberbotics/webots/blob/master/src/webots/engine/WbSimulationWorld.cpp#L150

This looks like a good place to improve the stability. Could you try optimizing that part of the code? Maybe reducing the size of the `mElapsedTimeHistory` list?

closed 2021-11-15 by BenjaminDeleze